stereolabs/zed-python-api #1: Error while building as per the instructions

royDFKI, 2017-08-07:
Hi,

Unfortunately I am unable to build the python wrappers for my project. Below is the error message:

D:\zed-python-master>python setup.py build
Compiling pyzed/defines.pyx because it changed.
[1/1] Cythonizing pyzed/defines.pyx
Compiling pyzed/types.pyx because it changed.
[1/1] Cythonizing pyzed/types.pyx
Compiling pyzed/core.pyx because it changed.
[1/1] Cythonizing pyzed/core.pyx
Compiling pyzed/mesh.pyx because it changed.
[1/1] Cythonizing pyzed/mesh.pyx
Compiling pyzed/camera.pyx because it changed.
[1/1] Cythonizing pyzed/camera.pyx
running build
running build_py
creating build
creating build\lib.win-amd64-3.5
creating build\lib.win-amd64-3.5\pyzed
copying pyzed\__init__.py -> build\lib.win-amd64-3.5\pyzed
copying pyzed\camera.pxd -> build\lib.win-amd64-3.5\pyzed
copying pyzed\core.pxd -> build\lib.win-amd64-3.5\pyzed
copying pyzed\defines.pxd -> build\lib.win-amd64-3.5\pyzed
copying pyzed\mesh.pxd -> build\lib.win-amd64-3.5\pyzed
copying pyzed\types.pxd -> build\lib.win-amd64-3.5\pyzed
running build_ext
building 'pyzed.defines' extension
creating build\temp.win-amd64-3.5
creating build\temp.win-amd64-3.5\Release
creating build\temp.win-amd64-3.5\Release\pyzed
C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\BIN\x86_amd64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -IC:\Users\Standard\AppData\Local\Programs\Python\Python35\lib\site-packages\numpy\core\include "-IC:\Program Files (x86)\ZED SDK\include" "-IC:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v8.0/include" -IC:\Users\Standard\AppData\Local\Programs\Python\Python35\include -IC:\Users\Standard\AppData\Local\Programs\Python\Python35\include "-IC:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\INCLUDE" "-IC:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\ATLMFC\INCLUDE" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.14393.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.14393.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.14393.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.14393.0\winrt" /EHsc /Tppyzed/defines.cpp /Fobuild\temp.win-amd64-3.5\Release\pyzed/defines.obj
defines.cpp
pyzed/defines.cpp(4675): error C2039: 'MEASURE_NORMALS': is not a member of 'sl'
C:\Program Files (x86)\ZED SDK\include\sl/defines.hpp(106): note: see declaration of 'sl'
pyzed/defines.cpp(4675): error C2065: 'MEASURE_NORMALS': undeclared identifier
pyzed/defines.cpp(4687): error C2039: 'MEASURE_DISPARITY_RIGHT': is not a member of 'sl'
C:\Program Files (x86)\ZED SDK\include\sl/defines.hpp(106): note: see declaration of 'sl'
pyzed/defines.cpp(4687): error C2065: 'MEASURE_DISPARITY_RIGHT': undeclared identifier
pyzed/defines.cpp(4699): error C2039: 'MEASURE_DEPTH_RIGHT': is not a member of 'sl'
C:\Program Files (x86)\ZED SDK\include\sl/defines.hpp(106): note: see declaration of 'sl'
pyzed/defines.cpp(4699): error C2065: 'MEASURE_DEPTH_RIGHT': undeclared identifier
pyzed/defines.cpp(4711): error C2039: 'MEASURE_XYZ_RIGHT': is not a member of 'sl'
C:\Program Files (x86)\ZED SDK\include\sl/defines.hpp(106): note: see declaration of 'sl'
pyzed/defines.cpp(4711): error C2065: 'MEASURE_XYZ_RIGHT': undeclared identifier
pyzed/defines.cpp(4723): error C2039: 'MEASURE_XYZRGBA_RIGHT': is not a member of 'sl'
C:\Program Files (x86)\ZED SDK\include\sl/defines.hpp(106): note: see declaration of 'sl'
pyzed/defines.cpp(4723): error C2065: 'MEASURE_XYZRGBA_RIGHT': undeclared identifier
pyzed/defines.cpp(4735): error C2039: 'MEASURE_XYZBGRA_RIGHT': is not a member of 'sl'
C:\Program Files (x86)\ZED SDK\include\sl/defines.hpp(106): note: see declaration of 'sl'
pyzed/defines.cpp(4735): error C2065: 'MEASURE_XYZBGRA_RIGHT': undeclared identifier
pyzed/defines.cpp(4747): error C2039: 'MEASURE_XYZARGB_RIGHT': is not a member of 'sl'
C:\Program Files (x86)\ZED SDK\include\sl/defines.hpp(106): note: see declaration of 'sl'
pyzed/defines.cpp(4747): error C2065: 'MEASURE_XYZARGB_RIGHT': undeclared identifier
pyzed/defines.cpp(4759): error C2039: 'MEASURE_XYZABGR_RIGHT': is not a member of 'sl'
C:\Program Files (x86)\ZED SDK\include\sl/defines.hpp(106): note: see declaration of 'sl'
pyzed/defines.cpp(4759): error C2065: 'MEASURE_XYZABGR_RIGHT': undeclared identifier
pyzed/defines.cpp(4771): error C2039: 'MEASURE_NORMALS_RIGHT': is not a member of 'sl'
C:\Program Files (x86)\ZED SDK\include\sl/defines.hpp(106): note: see declaration of 'sl'
pyzed/defines.cpp(4771): error C2065: 'MEASURE_NORMALS_RIGHT': undeclared identifier
pyzed/defines.cpp(4964): error C2039: 'VIEW_NORMALS': is not a member of 'sl'
C:\Program Files (x86)\ZED SDK\include\sl/defines.hpp(106): note: see declaration of 'sl'
pyzed/defines.cpp(4964): error C2065: 'VIEW_NORMALS': undeclared identifier
pyzed/defines.cpp(4976): error C2039: 'VIEW_DEPTH_RIGHT': is not a member of 'sl'
C:\Program Files (x86)\ZED SDK\include\sl/defines.hpp(106): note: see declaration of 'sl'
pyzed/defines.cpp(4976): error C2065: 'VIEW_DEPTH_RIGHT': undeclared identifier
pyzed/defines.cpp(4988): error C2039: 'VIEW_NORMALS_RIGHT': is not a member of 'sl'
C:\Program Files (x86)\ZED SDK\include\sl/defines.hpp(106): note: see declaration of 'sl'
pyzed/defines.cpp(4988): error C2065: 'VIEW_NORMALS_RIGHT': undeclared identifier
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio 14.0\\VC\\BIN\\x86_amd64\\cl.exe' failed with exit status 2

Help regarding this would really be appreciated.

Thanks,
Roy.

adujardin, 2017-08-08:
Hi,

Which version of the ZED SDK are you using ? 
This wrapper needs the latest [2.1](https://www.stereolabs.com/developers/release/2.1/), it seems you have the 2.0 installed.

royDFKI, 2017-08-08:
Yeah, it did solve the issue.
Thanks.

closed 2017-08-08 by royDFKI