stereolabs/zed-python-api #2: error on building the zed sdk for python

LearningVision, 2017-08-08:
I have tried to open the stereo video .svo file in python.
I have download the zed-sdk 2.1(for windows 64x) and then tried to build the zed-python package.
But I got error like as follow.
I think I installed the SDK and I can display the .svo file with zed explorer.exe

E:\work\SVO\zed-python>python setup.py install
Error: you must install the ZED SDK.
Compiling pyzed/defines.pyx because it changed.
[1/1] Cythonizing pyzed/defines.pyx
Compiling pyzed/types.pyx because it changed.
[1/1] Cythonizing pyzed/types.pyx
Compiling pyzed/core.pyx because it changed.
[1/1] Cythonizing pyzed/core.pyx

Error compiling Cython file:
------------------------------------------------------------
...
        bool save(types.String filename)
        bool load(types.String filename)

    cdef cppclass SpatialMappingParameters 'sl::SpatialMappingParameters':
        ctypedef pair[float, float] interval
       ^
------------------------------------------------------------

pyzed\camera.pxd:117:8: Expected an identifier, found 'ctypedef'

Error compiling Cython file:
------------------------------------------------------------
...
        bool save(types.String filename)
        bool load(types.String filename)

    cdef cppclass SpatialMappingParameters 'sl::SpatialMappingParameters':
        ctypedef pair[float, float] interval
                                   ^
------------------------------------------------------------

pyzed\camera.pxd:117:36: Syntax error in C variable declaration

Error compiling Cython file:
------------------------------------------------------------
...
    def right_cam(self):
        return self.py_right_cam

cdef class PyCameraInformation:
    def __cinit__(self, camera.PyZEDCamera py_camera, PyResolution resizer):
                       ^
------------------------------------------------------------

pyzed\core.pyx:162:24: 'PyZEDCamera' is not a type identifier

Error compiling Cython file:
------------------------------------------------------------
...

cdef class PyCameraInformation:
    def __cinit__(self, camera.PyZEDCamera py_camera, PyResolution resizer):
        res = Resolution(resizer.width, resizer.height)
        self.py_calib = PyCalibrationParameters()
        self.py_calib.calibration = py_camera.camera.getCameraInformation(res).calibration_parameters
                                                                             ^
------------------------------------------------------------

pyzed\core.pyx:165:78: Cannot convert Python object to 'CalibrationParameters'

Error compiling Cython file:
------------------------------------------------------------
...
    def __cinit__(self, camera.PyZEDCamera py_camera, PyResolution resizer):
        res = Resolution(resizer.width, resizer.height)
        self.py_calib = PyCalibrationParameters()
        self.py_calib.calibration = py_camera.camera.getCameraInformation(res).calibration_parameters
        self.py_calib_raw = PyCalibrationParameters()
        self.py_calib_raw.calibration = py_camera.camera.getCameraInformation(res).calibration_parameters_raw
                                                                                 ^
------------------------------------------------------------

pyzed\core.pyx:167:82: Cannot convert Python object to 'CalibrationParameters'
Traceback (most recent call last):
  File "setup.py", line 149, in <module>
    extList = cythonize(create_extension(mod[0], mod[1]), compiler_directives=cython_directives)
  File "C:\Anaconda3\lib\site-packages\Cython\Build\Dependencies.py", line 915, in cythonize
    cythonize_one(*args)
  File "C:\Anaconda3\lib\site-packages\Cython\Build\Dependencies.py", line 1037, in cythonize_one
    raise CompileError(None, pyx_file)
Cython.Compiler.Errors.CompileError: pyzed/core.pyx

adujardin, 2017-08-08:
Hi,
It seems that the ZED SDK is not detected since you have "`Error: you must install the ZED SDK`". Try reinstalling it and reboot. 
Then you can also test the C++ samples and the other tools (ZEDFu or ZED Depth Viewer).

ogig3, 2017-09-08:
Hi,
I got a same error.
This will be caused because cython version is old. please check your python version by 

```
pip freeze
```

and if your python is not 0.26, update your cython by

```python
pip install cython==0.26
```

closed 2018-09-13 by adujardin