stereolabs/zed-python-api #3: Point Cloud data access

royDFKI, 2017-08-16:
Hi,
I am trying to manipulate some robots using the zed camera and due to this I need the XYZ coordinates with respect to the camera. I am able to retrieve it using the retrieve_measure function but since multiple robots are functioning simultaneouls I need the data to be shared between different processes.

But, unfortunately, the data is not flowing in properly, so I though I would retrieve the data from this variable and store it as a numpy array and share it. My problem is that I am unable to understand the difference between the get_data and get_value_functions i.e. 
>>> point_cloud.get_data()[600][400]
array([172, 191,  15, 197], dtype=uint8)
>>>point_cloud.get_value(600,400)
(Error code:  Success, array([ -232.36811829,   226.42419434,  3275.38720703,            nan]))
The pixel values has to be given as arguments.
What do the get_data and get_value signify, since there is no documentation about get_data fnction in the Mat class of the API documentation?

Regards,
Roy.

nesnes, 2017-08-16:
Hi,
If you want to get the XYZ values at an x,y position, you should call point_cloud.get_value(x,y)

royDFKI, 2017-08-16:
Alright, I understand now, but it there a way to transfer the get_value function data for all the pixels to a numpy array? Or some other way that this point_cloud can be shared between two threads in python?

closed 2022-04-27 by github-actions[bot]

PINTO0309, 2024-04-20:
This issue is already closed, but since it did not seem to provide an accurate answer to the intent of the questioner's question, I will describe the solution I found, just in case.

```
pyzed==4.0
```

`point_cloud.get_value(x, y)` can only get the distance of one arbitrary point, but sometimes we want to get the distance of all the points in the angle of view at once.

```python
zed = sl.Camera()

# Create a InitParameters object and set configuration parameters
init_params = sl.InitParameters()
init_params.depth_mode = sl.DEPTH_MODE.ULTRA  # Use ULTRA depth mode
init_params.coordinate_units = sl.UNIT.MILLIMETER  # Use meter units (for depth measurements)

# Open the camera
status = zed.open(init_params)
point_cloud = sl.Mat()

zed.retrieve_measure(point_cloud, sl.MEASURE.XYZRGBA)
print(point_cloud.get_data().shape) # [720, 1280, 4]
print(point_cloud.get_data().dtype) # dtype('float32')

# Simplified logic that does not take into account that `Inf`.
# Distance values for all coordinates in 
# `math.isfinite(point_cloud.get_data()[:, :, 2]) == Flase` are invalid values
distances = np.linalg.norm(point_cloud.get_data()[:, :, :3], axis=2)
print(distances.shape) # (720, 1280)

# Obtain the distance to the center of the screen
print(distances[360, 640]) # 4206.3887 (mm)
```