stereolabs/zed-python-api #4: ImportError: No module named camera

rex9999772001ML, 2017-08-18:
Hello, thanks for the project. While I try to execute example and tutorial code, **all of them prompt same kind of error**. Take positional_tracking as an example: 
`Traceback (most recent call last):`
 `File "positional_tracking.py", line 21, in <module>`
 `    import pyzed.camera as zcam`
`ImportError: No module named camera`

I doubt it is because of different python version(I use python 2.7 and this project require me to use python 3.5+). If it is the reason, I would like to ask is there any method to solve this issue without updating python? If it is not, how could I solve this issue?

Thank you

RSKothari, 2017-08-29:
I have the same error. I don't think it's because of the python version because I'm using Python 3.5+. I tried searching for all available modules under Pyzed and it returns empty. I'm hoping for a fix. I also did receive a lot of warnings while building zed-python, not sure if they are related but 
`sudo python3 setup.py build`
returns

>  #warning "Using deprecated NumPy API, disable it by " "#defining NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION

Upon further inspection, it seems this warning can be safely ignored.

rex9999772001ML, 2017-09-04:
Hello @RSKothari, My problem was solved after I update the latest ZED SDK with python 3.5. Maybe you could try this method and see if you could find camera module.

prattcmp, 2017-10-18:
@rex9999772001ML Could you go into more detail about how you resolved this issue? I am having the same problem.

RSKothari, 2017-10-18:
@rex9999772001ML my issue was slightly more complicated. Python, by default, used 2.7. For instance, if I typed <python --version> into my terminal it used to point towards 2.7. I had to change what <python> meant. I then changed it such that running <python> ran <python3>. Post that, it all worked smoothly for me.
Pro tip: python --version and sudo python --version are two completely different configurations. If you are using python, please double check which version <sudo> reverts to.

chan-e91, 2017-10-24:
Hi RSKothari, I am facing the same issue.
How to run this file with python3,
I am using opencv virtual environment with python2.7.

RSKothari, 2017-10-24:
Hi @chan-e91 , zed-python is only available in python3. First, check the following in terminal please:
python --version
sudo python --version

If they both return 3.5, then proceed ahead with the steps given in the README. If not, then you'll need to change the default python to another alternative. Following the steps here should solve the problem.

https://linuxconfig.org/how-to-change-from-default-to-alternative-python-version-on-debian-linux

chan-e91, 2017-10-25:
I am using opencv in virtual environment,
in that vm, python --version returns 3.5.2, but sudo python --version gives 2.7.12.
I once did change the python's alias .. but it wreaked havoc, and i have todelete and reinstall the opencv and python.

chan-e91, 2017-10-25:
I was able to resolve issue, in python install, it looks for cuda-8.0 as default cuda, after assigning correct cuda path( i am using cuda9) it resolved.
but, a new error crept up ... "     i**mport pyzed.camera as zcam
ImportError: libcudart.so.8.0: cannot open shared object file: No such file or directory**" 
Did anyone face this error, i think by default it takes cuda 8

p0wdrdotcom, 2017-12-18:
+1

It shouldn't be too much effort to make a python 2.7 compatible version.

viniciusarruda, 2018-04-18:
Anyone solved this issue ? 
```
import pyzed
print(dir(pyzed))
# gives me ['__builtins__', '__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__path__', '__spec__']
```

closed 2018-09-13 by adujardin