stereolabs/zed-python-api #6: Using zed-python on the Jetson TX1 platform conflicts with OpenCV version

axjensen, 2017-09-12:
Hi,
I would like zed-python on the Jetson TX1 platform, but I can't figure out how.
Jetson TX1 can come with its own optimized OpenCV libraries, and this might one of my problems.

I build the zed-python plugin with Python3.5, which seems to go fine.
I can run the 'hello_zed.py' example and I get the serial number, so everything seems fine.
But, when I try to use the 'live_camera.py'example, python3.5 cant find the cv2 module (the OpenCV python binding).
If I use python2, I can use the cv2 module, but now I can't use the zed-python plug in.
What would be your recommendation on using zed-python plugin, OpenCV and JTX1?
Do I stick to Python2 (and try to build zed-python by Python2)?
Will zed-python work with Python2 (requirements says Python3.5)?
Can I stick with libopencv4tegra / libopencv4tegra-python ? Or do I need to build OpenCV from source ( and loose optimization)?

adujardin, 2017-09-12:
Hi,
You won't be able to compile zed-python with Python 2. 
In your case, you should probably build a recent version of OpenCV (3.3) from source which includes some (if not most, according to the [changelog)](https://github.com/opencv/opencv/wiki/ChangeLog) of the Tegra optimisations, with the python binding module.

axjensen, 2017-09-12:
Thanks @adujardin . I will try that.

closed 2017-09-12 by axjensen

axjensen, 2017-09-13:
Hi @adujardin 
I followed your recommendation and installed OpenCV 3.3.
I can now also run the 'live_camera.py' example.
Thank you for your help.

chan-e91, 2017-12-14:
Hi, 
I am also trying to install this in TX1, I have installed opencv 3.3 in a virtual environment.
But when i try to install ZED SDK version 2.2, it opens a webpage, containing script 
"https://cdn.stereolabs.com/developers/downloads/ZED_SDK_Linux_JTX1_v2.2.1.run",  how to download and run from this script ?
thanks in advance :)