stereolabs/zed-python-api #7: error on jetson tx2

emmanuelgeoffray, 2017-09-15:
Hello,

Did you succeed to install `zed-python` on Jetson TX2?
When running `python3 setup.py build`, I get many errors like this one:
`error: using typedef-name ‘sl::MeshFilterParameters::FILTER’ after ‘enum’`

* gcc: 5.4.0
* python: 3.5.2
* cython: 0.26.1 (tried 0.26.0 but same error)
* numpy: 1.13.1
* Ubuntu 16.04

Thanks for your help,
Emmanuel

adujardin, 2017-09-15:
Hi,
Which ZED SDK version are you using?
zed-python is meant to be used with the latest version ([2.1](https://www.stereolabs.com/developers/release/2.1/))

emmanuelgeoffray, 2017-09-15:
sorry, I forgot to mention that.
zed sdk version is 2.1.2

emmanuelgeoffray, 2017-09-18:
Here is the full log.
The python package has been tested on an arm processor?
If yes, I will try to reinstall the jetson from scratch.

[zed-python-build.txt](https://github.com/stereolabs/zed-python/files/1310723/zed-python-build.txt)

adujardin, 2017-09-25:
This package wasn't tested on the jetson yet, unfortunately.
I very briefly tested it and it seems there is an issue since I got very similar errors.
I will try looking into it next week.

emmanuelgeoffray, 2017-09-25:
thanks for the update
best,
e

commit 546aef63 (2017-10-02): Fix #7 Compilation issue Tegra
closed 2017-10-02 by adujardin