stereolabs/zed-python-api #8: svo_real_time_mode not working

fperucic, 2017-09-25:
When `svo_real_time_mode=True` is set on the `zcam.PyInitParameters` the main grab loop will be unable to fetch frames.

If set to `false`, everything will work. However the playback will not be in real time.

adujardin, 2017-09-25:
Hi,
I just tested it with multiple SVO and everything works as expected, with or without `svo_real_time_mode`.

Could you try with this SVO? https://www.stereolabs.com/download_327af3/SVO_samples/sofa.svo

fperucic, 2017-09-26:
Yes, it works with your SVO. Maybe the reason was the LOSSLESS compression used on my SVO files.

closed 2017-09-26 by fperucic

adujardin, 2017-09-26:
It may be a bug with the recording or something. The LOSSLESS compression doesn't work with real-time SVO reading but the expected behavior is to ignore the parameter and still read the images (just not in real time). 

Would that be possible to share one of the SVO you recorded that caused this issue, please?

fperucic, 2017-10-02:
https://transfer.sh/suhJC/1f439ce0-8412-11e7-9364-d46e0e1fc4ba_12768_done.svo

I think this is the one

adujardin, 2017-10-02:
Thanks ! It will be useful. I noticed a bug triggered by this SVO which look similar to yours even with the C++ API.