stereolabs/zed-python-api #11: Import Error : Libcudaart.8.0 not found, when installed on ubuntu 16.04 - CUDA 9.0

chan-e91, 2017-10-25:
Hi,
I have installed CUDA 9.0 , and built the python setup file, by pointing to cuda installed path.(/usr/local/CUDA 9.0/ )
The build was successful, but whn i try to run live_camer.py, following throws up : 
**" Traceback (most recent call last):
  File "live_camera.py", line 27, in <module>
    import pyzed.camera as zcam
ImportError: libcudart.so.8.0: cannot open shared object file: No such file or directory "**

I think it is related to CUDA 8 being taken as default cuda version,
 can you please check into this, and give some alternative. Thanks in advance

adujardin, 2017-10-25:
Hi,
Each ZED SDK is tied to a specific CUDA version. In this case CUDA 8 for ZED SDK 2.1. You can install 2 CUDA versions alongside if you need CUDA 9.

chan-e91, 2017-10-25:
Should i install CUDA 8 now, alongside, CUDA 9 ?

adujardin, 2017-10-25:
Yes, or uninstall CUDA 9 and keep only CUDA 8 if you don't need 9

chan-e91, 2017-10-26:
thanks, I was able to run after i have have installed cuda 8.

closed 2017-10-26 by chan-e91