stereolabs/zed-python-api #12: Only left Zed camera is displaying image.

chan-e91, 2017-10-30:
Hi,
 I was able to capture images only from the left camera. 
I tried to change the "PyVIEW.PyVIEW_LEFT" to "PyVIEW.PyVIEW_RIGHT" in camera.pyx file but to no use ( as VIEW_RIGHT was defined ), can you please tell me how to take 2 images, and rectify them using this wrapper .

adujardin, 2017-10-30:
Hi,
To get the right image in the example you can replace [this line](https://github.com/stereolabs/zed-python/blob/master/examples/live_camera.py#L58)
`cam.retrieve_image(mat)` by `cam.retrieve_image(mat, sl.PyVIEW.PyVIEW_RIGHT)`

You should not change directly the wrapper because you're only changing the default value.

chan-e91, 2017-10-30:
thanks for the quick reply, the images we get from this are rectified ?

adujardin, 2017-10-30:
Yes, to get the unrectified images call it with `VIEW_LEFT_UNRECTIFIED` (the complete list is [here](https://github.com/stereolabs/zed-python/blob/master/pyzed/defines.pxd#L112))

chan-e91, 2017-10-30:
yes, I needed only rectified images. Thanks for the help.!!

closed 2017-10-30 by chan-e91