stereolabs/zed-python-api #15: Can't build plugin, tried all suggestions

rileysbudd, 2017-11-10:
I've been trying to build the python wrapper for the ZED SDK. I've seen other people have similar issues and tried their solutions, but haven't made any progress. Regardless of what I do I get the "Error: you must install the ZED SDK."

Here are the things I've tried:

checked Cython >=.26
checked python.exe is the 64 bit version
uninstalled and reinstalled latest SDK twice

I originally downloaded the plugin to my project folder, so I also tried to:
move location of zed-python to ...\plugins folder of SDK
move location of zed-python to ...\Anaconda3\Lib\site-packages

And every time I ran setup.py, from wherever I ran it, it didn't think that the ZED SDK was installed.

Any ideas? The SDK tools for this camera are cool, but if I can't use the python wrapper it's sort of just an expensive toy...

SBrougham, 2017-11-12:
I am having the same issue, I tried an older version of the SDK and it still didn't think it was installed.

mpowelson, 2017-12-08:
Just to add to the list, I also am facing this problem. I am running Python 3.6 through Anaconda on Windows 10 with the latest ZED SDK (2.2.1). If anyone finds a solution to this, please let us know.

crross0827, 2017-12-13:
I am also having issues using python 3.6 through Anaconda on Windows 10 with ZED SDK (2.2.1). I fixed the  "Error: you must install the ZED SDK." error by creating an environment variable called ZED_INCLUDE_DIRS that pointed to the include folder in the ZED SDK folder, "C:\Program Files (x86)\ZED SDK\include". Now I'm running into an error originating from the _msvccompiler.py file in distutils. The error is "line 551, in library_dir_option return "/LIBPATH:" + dir   TypeError: must be str, not NoneType". Does anyone know how to fix this?

crross0827, 2017-12-13:
To follow up, I tried this fix to my problem: https://bugs.python.org/issue13765 

I get another error stating that dir is NoneType not an iterable.

sjwhhhi, 2017-12-21:
Have anyone sloved this question?
I met the problem too.

mpowelson, 2018-01-04:
I believe I just fixed this on my Windows 10 Anaconda install. I needed to define the environment variables, as others have noted, inside my conda environment. Here's a paraphrase of an applicable stackoverflow question.

> With conda you can create these files:` /<path_to_env>/etc/conda/activate.d/env_vars.bat` which will be executes when the environment is activated. Put something like this in it:
> ```
> set ZED_LIBRARY_DIR = C:\Program Files (x86)\ZED SDK\lib
> set ZED_INCLUDE_DIRS = C:\Program Files (x86)\ZED SDK\include
> ```
> Similarly `/<path_to_env>/etc/conda/deactivate.d/env_vars.bat `gets executed when the env is deactivated.set
> ```
> ZED_LIBRARY_DIR = 
> ZED_INCLUDE_DIRS = 
> ```
> 
> (On Windows you delete env-vars by setting them to nothing.)

https://stackoverflow.com/questions/46048210/environment-variables-on-windows-in-virtualenv-virtualenvwrapper-win

closed 2018-09-13 by adujardin