stereolabs/zed-python-api #16: The speed of depth sensing (zed.retrieve_measure)

lihp6207, 2017-11-21:
I have a problem about processing speed of function(zed.retrieve_measure)
The FPS is unstable and so low , from 4 to 20.
The following is my setting:

init_params.depth_mode = sl.PyDEPTH_MODE.PyDEPTH_MODE_PERFORMANCE
init_params.coordinate_units = sl.PyUNIT.PyUNIT_METER  
init_params.camera_resolution = sl.PyRESOLUTION.PyRESOLUTION_HD720   
runtime_parameters.sensing_mode = sl.PySENSING_MODE.PySENSING_MODE_STANDARD  

Then, I measure the time of the following functions cost is unstable and so long(0.05 to 0.3 second):
            # Retrieve left image
            zed.retrieve_image(image, sl.PyVIEW.PyVIEW_LEFT)
            # Retrieve depth map. Depth is aligned on the left image
            zed.retrieve_measure(depth, sl.PyMEASURE.PyMEASURE_DEPTH)
            # Retrieve colored point cloud. Point cloud is aligned on the left image.
            zed.retrieve_measure(point_cloud, sl.PyMEASURE.PyMEASURE_XYZRGBA)

Third, I open the ZED Depth Viewer using the same device. It runs smoothly and the FPS is above 30.

Would someone answer the questions, thank you very much.
BTW, the environment is based on TX2 ubuntu16.04 and python3.5.

RSKothari, 2017-11-21:
@lihp6207 Could you try the following? Comment out one of the data acquisitions and view the change in performance. Also, could it be that you were using USB 2.0 and while running on ZED Depth Viewer, on USB 3.0? Just a thought.

lihp6207, 2017-11-22:
Dear @RSKothari  :
Thanks for your responding!!
I found the problem why delay so much. The chrome explorer caused acquiring delay. Also I tried to retrieve point cloud only. It still time-consuming. The FPS is around 10 to 12. It is also slower than acquiring from ZED depth Viewer(30FPS above).
Then, I've reconfirmed that it's connected by USB3.0 between ZED and TX2.

adujardin, 2018-09-13:
Hi,
The python wrapper adds a layer of computation/conversion that can significantly reduce the performance and add latency compared to the native C++ SDK, especially when compared to a GPU implementation such as Depth Viewer.

The performances are an important consideration of the wrapper implementation but that's the cost of using python, unfortunately there's not much else we can do.

For critical real-time applications, the C++/CUDA SDK should be considered instead.

closed 2018-09-13 by adujardin