stereolabs/zed-python-api #20: Error: you must install the ZED SDK.

sjwhhhi, 2017-12-21:
When I run setup.py build, I got this error.
However, I have installed ZED SDK for severl times. I am sure I installed it successfully because I can run the examples.
Anyon have some ideas? Thanks!

sjwhhhi, 2017-12-21:
> C:\zed>python setup.py build
Error: you must install the ZED SDK.
Compiling pyzed/defines.pyx because it changed.
[1/1] Cythonizing pyzed/defines.pyx
Compiling pyzed/types.pyx because it changed.
[1/1] Cythonizing pyzed/types.pyx
Compiling pyzed/core.pyx because it changed.
[1/1] Cythonizing pyzed/core.pyx
Compiling pyzed/mesh.pyx because it changed.
[1/1] Cythonizing pyzed/mesh.pyx
Compiling pyzed/camera.pyx because it changed.
[1/1] Cythonizing pyzed/camera.pyx
running build
running build_py
running build_ext
building 'pyzed.defines' extension
C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\BIN\x86_amd64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -IC:\Users\sjwhh\AppData\Local\Programs\Python\Python36\include -IC:\Users\sjwhh\AppData\Local\Programs\Python\Python36\include "-IC:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\INCLUDE" "-IC:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\ATLMFC\INCLUDE" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.14393.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.14393.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.14393.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.14393.0\winrt" /EHsc /Tppyzed/defines.cpp /Fobuild\temp.win-amd64-3.6\Release\pyzed/defines.obj
defines.cpp
pyzed/defines.cpp(507): fatal error C1083: 无法打开包括文件: “sl/types.hpp”: No such file or directory
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio 14.0\\VC\\BIN\\x86_amd64\\cl.exe' failed with exit status 2

Alonmeytal, 2017-12-27:
@sjwhhhi just got it myself. The setup.py is looking for an environment variable called ZED_INCLUDE_DIRS while the installation creates one called ZED_SDK_ROOT_DIR.
Short fix is to create it yourself and set it to where the "include" folder inside where ZEN SDK was installed for me it was C:\Program Files (x86)\ZED SDK\include

adambeytin, 2017-12-27:
I also had this problem. Not only did you have to add ZED_INCLUDE_DIRS to your environment variables and point it to the include directory, but also ZED_LIBRARY_DIR pointed to the lib directory. Both are just the ZED_SDK_ROOT_DIR plus \include or \lib.

closed 2018-09-13 by adujardin