stereolabs/zed-python-api #21: Can't use with TensorFlow

adambeytin, 2017-12-27:
I would like to be able to use the ZED camera with the TensorFlow libraries in Python but cannot with zed-python because of version support mismatch.

ZED SDK 2.2.1 only supports CUDA 9, though 2.2.0 supports CUDA 8.
TensorFlow only supports CUDA 8.

zed-python can be built with ZED SDK 2.2.0, but when I try to open a pyzed.camera I get the following error:

> ZED (Init) >> ERROR: Executable was designed for a newer version of the SDK.
EXE v2.2.1 / installed v2.2.0

Is there any solution?

orenphysi, 2018-01-08:
This issue is critical for me as well, please advise...

nesnes, 2018-01-10:
> ZED (Init) >> ERROR: Executable was designed for a newer version of the SDK. EXE __v2.2.1__ / installed __v2.2.0__

This error is thrown when a program compiled with a higher version of the SDK is executed with a lower one installed. Just like you are still using the zed-python binaries compiled with the 2.2.1 but with the SDK 2.2.0.

I'll recommend you to clear the current installation of the ZED python wrapper, install the ZED SDK 2.2.0, fresh clone the zed-python repo and follow the compilation and installation instructions.

imgntn, 2018-01-13:
TF 1.5 now supports CUDA 9 https://github.com/tensorflow/tensorflow/releases

adambeytin, 2018-01-14:
Thank you very much, removing the zed-python wrapper completely, cloning it again, and recompiling fixed the problem. I had assumed that the cleanall function in setup.py would accomplish the same thing.

That's great news about TF 1.5 supporting CUDA 9! Thank you for the update.

closed 2018-01-14 by adambeytin