stereolabs/zed-python-api #22: Saving a loading Depth data from FS

Alonmeytal, 2018-01-09:
I am using the ZED as input for MV algorithm.
I first analyze the RGB image, and then (after quite a while) I get back X, Y values for which I need the Z value (from the depth/point cloud).
What would be the best way to do that? Moreover, if I want to reprocess my data, do I have a way to get the Z value retroactively from a saved depth/point-cloud file, and if so, how would I save my Mat to the FS so that I can retrieve it later?

Thanks!

Alonmeytal, 2018-01-16:
@nesnes @adujardin

nesnes, 2018-01-17:
You can save the color image and the depth one in a file, using OpenCV for example, but we won't recommend this solution.

The ZED SDK offers the SVO file format to record a sequence from the ZED. After recording one you can play it back in the SDK to fully simulate a ZED.
You can record a sequence from the ZED Explorer tool, or from the SDK.
To play it back, just specify its path in the initParameters: https://github.com/stereolabs/zed-python/blob/master/examples/read_svo.py#L42

Alonmeytal, 2018-01-17:
@nesnes thanks for the response, but I think you didn't understand my use case:
My app provides a view & record functionality (just like the Zed Viewer), as my users would like to watch their subject "live". We'd also like to use the frames coming on the go for some ML processing, further complicating matters.

nesnes, 2018-01-17:
From what you said, you will generate many sequences with the ZED, extract the images and the depth, save them on your hard drive, and give them as an input to your machine learning algorithm.

In these conditions, using an SVO file that requires being opened by the ZED SDK won't be the best solution.
I'll suggest you to take a look at the SVO Export sample that turns SVOs into image and depth sequences in 16-bit png format. It uses OpenCV.
https://github.com/stereolabs/zed-examples/tree/master/svo%20recording/export

This should be a great tool to generate an ML dataset from your SVO files.

Alonmeytal, 2018-01-17:
@nesnes if you could answer 2 "follow-up questions":
1. In the ZED Viewer: How do you retrieve the image for preview (on-screen) while also recording it to SVO?
2. I understand that saving the video stream from both cameras allow the SDK to create the depth reading from the saved data (as if it was "fresh" data coming in), but is there any transformation that would allow me to keep the result of this calculation? Will using depth.get_data() get me a matrix of depth values aligned to one of the cameras?

Thanks!

closed 2022-04-27 by github-actions[bot]