stereolabs/zed-python-api #23: Could I retrieve depth by images loaded from files?

Hydraz320, 2018-01-17:
I know I can get real-time images and depth, however, I have recorded videos and could I get depth from video data which includes left and right frames?

Thanks a lot!

Alonmeytal, 2018-01-21:
I am pretty sure that if you've used the internal record function (to SVO), you can load it back to memory like in the read-svo example, and extract depth retroactively :
https://github.com/stereolabs/zed-python/blob/master/examples/read_svo.py

Hydraz320, 2018-01-29:
Thank you for your response! I mean if I could get images with ZED connected to my Mac and then put images on the server with GPU to calculate depth?

Alonmeytal, 2018-01-29:
I'd guess that running the ZED package on a server and feeding it with your SVO might not be enough, but that's something the ZED should confirm.

obraun-sl, 2018-01-29:
Hi,

That's not possible for the moment to send images through network  and get them straight on the ZED SDK. The SVO needs to run on the same platform that the GPU. 

What you can do is record the SVO on a different PC (but mac is not compatible btw), then send the SVO to the server with p2p communication, and once the svo is fully written on the server, you can run the zed sdk on that server with the SVO file as input. Not tested, but should be possible.

/*OB*/

closed 2018-09-13 by adujardin