stereolabs/zed-python-api #26: Found CUDA 9.0 but the ZED SDK requires 8.0

klnsz, 2018-01-25:
Hi. I am on jetson tx2 with jetpack 3.2. This version of jetpack comes with cuda 9 but  when i try to install zed sdk 2.3 version i got an error saying that "Found CUDA 9.0 but the ZED SDK requires 8.0". Isn't this version of zed sdk comes with a support of cuda 9 or am i missing something?

adujardin, 2018-01-25:
Hi,
The ZED SDK 2.3.0 doesn't currently support the jetpack 3.2 (still in developer preview), and therefore only support Cuda 8 on jetson (jetpack 3.1). 

However, we plan to release an update for this (probably ZED SDK 2.3.1).

sethmnielsen, 2018-02-05:
I noticed that on the stereolabs website there is now an option to download "ZED SDK for Jetson TX2 v2.3 (Jetpack 3.2, CUDA 9)". I just tried installing the SDK on my TX2 using Jetpack 3.2 and the installation completed successfully.

However, when I try to run any of the ZED tools, I get the error:
`error while loading shared libraries: libzstd.so.0: cannot open shared object file: No such file or directory
`

Doing a quick search, I found libzstd.so.1 and libzstd.so under /usr/local/lib, but not libzstd.so.0. 
I'm not sure if this is an issue with my setup or an issue with the newest ZED SDK release.

adujardin, 2018-02-05:
Hi,
Yes I forgot to mention that we released the ZED SDK 2.3.0 for the Jetpack 3.2 (Cuda 9)

@sethmnielsen for your issue, a simple `sudo apt install libzstd0` should fix the problem. There might be a small bug in the installation script.

adujardin, 2018-02-06:
@sethmnielsen we released a fix for this issue, [ZED SDK 2.3.0b for Jetpack 3.2](https://download.stereolabs.com/zedsdk/2.3/tegrax2_jp32)

sethmnielsen, 2018-02-07:
Beautiful. The fix worked! I have the ZED running on a TX2 with Jetpack 3.2 and CUDA 9.0.

Thanks you so much for the extremely fast response and fix! Stereolabs rocks!

avirtuos, 2018-03-23:
Doesn't look like you posted this fix for the TX1. Where can I find it?

adujardin, 2018-03-23:
@avirtuos The support for the new Jetpack 3.2 for TX1 will be included in the upcoming version of the ZED SDK (2.4.0).

closed 2018-03-23 by adujardin

avirtuos, 2018-03-23:
Which will be released when? Even a rough ETA will help me understand if o should go find an alternative for my short term needs. Thanks.

adujardin, 2018-03-23:
It should be released in a few weeks.

mfilipen, 2018-04-03:
we are looking forward to release, thanks.

croadfeldt, 2018-04-09:
Any updates on when the ZED SDK 2.4.0 is due out?

joshuayuan, 2018-04-12:
I am also running into this original issue, and am excited for the upcoming update!

Hunterhal, 2018-04-13:
Hello having same issue with TX1, waiting for fix. Thanks in advance.

adujardin, 2018-04-17:
Hi, 
Thanks for your patience, here's a link to an RC version of the ZED SDK 2.4.0: http://download.stereolabs.com/zedsdk/2.4/RC

Please keep in mind that it's still an RC version and therefore may contain bugs. Feel free to contact us if you find something at support@stereolabs.com