stereolabs/zed-python-api #29: Cannot get depth image using OpenCV API

shuuchen, 2018-02-12:
Hi there,

The following code can read images in a light manner. But it cannot get depth image.
I just wonder whether we can get the depth image using OpenCV APIs.

It is part of https://github.com/stereolabs/zed-python/issues/28, but I opened a new
issue since it is a different topic.

```python
import numpy as np
import cv2

cap = cv2.VideoCapture(0) # depending on the ZED ID

while(True):
    ret, frame = cap.read() # Capture SbS frames
    height, width = frame.shape[:2]
    left = frame[0:height,0:int(width*0.5)] # Extract the left img

    # object detection
    .......

# Release the capture
cap.release()
cv2.destroyAllWindows()
```

Thanks very much.

obraun-sl, 2018-02-12:
Hi,

You need to use the ZED SDK if you want to get the depth map. The depth is not computed in the camera but in the ZED SDK. 
The only output you have from the camera is the left and right images.

Best,
/*OB*/

praneethshub, 2018-02-12:
@obraun-sl i also face same issue. Not able to find way to use ZED SDK functions after we are initiating ZED camera with Opencv-python line "cap = cv2.VideoCapture(0)"

Getting error when opening another instance of camera in same code with"zed = zcam.PyZEDCamera()"  for getting depth  with ZED-Python API. 

Want to know if any other way to use these depth functions from ZED-Python API, after receiving ZED images using opencv-python.

closed 2018-09-13 by adujardin