stereolabs/zed-python-api #30: "Error: Unable to find vcvarsall.bat" while building 'pyzed.defines' (Running the setup.py on windows)

Bauernteams, 2018-02-22:
Hello, 
As mentioned in the title i am trying to run the setup.py and get the error: unable to find vcvarsall.bat. Find the whole log below.

I have already checked/tried/done the following:
- installed/updated cython
- installed visual studio 2017

thank you very much in advance
and kind regards

-----------------------------------------------------------------------------------------------------------

C:\Users\wdeti\Desktop\ZED\zed-python>python setup.py build
ZED SDK Version: OK
compilation flags:
include dirs: ['Q:\\Programs\\Python\\Python36\\lib\\site-packages\\numpy\\core\\include', 'C:\\Program Files (x86)\\ZED SDK/include', 'Q:\\NVIDIA\\CUDA v9.1\\Development/include']
library dirs: ['Q:\\Programs\\Python\\Python36\\lib\\site-packages\\numpy\\core\\include', 'C:\\Program Files (x86)\\ZED SDK/lib', 'Q:\\NVIDIA\\CUDA v9.1\\Development/lib/x64']
libraries: ['sl_core64', 'sl_zed64']
Building module: ('pyzed.defines', ['pyzed/defines.pyx'])
Compiling pyzed/defines.pyx because it changed.
[1/1] Cythonizing pyzed/defines.pyx
Building module: ('pyzed.types', ['pyzed/types.pyx'])
Compiling pyzed/types.pyx because it changed.
[1/1] Cythonizing pyzed/types.pyx
Building module: ('pyzed.core', ['pyzed/core.pyx'])
Compiling pyzed/core.pyx because it changed.
[1/1] Cythonizing pyzed/core.pyx
Building module: ('pyzed.mesh', ['pyzed/mesh.pyx'])
Compiling pyzed/mesh.pyx because it changed.
[1/1] Cythonizing pyzed/mesh.pyx
Building module: ('pyzed.camera', ['pyzed/camera.pyx'])
Compiling pyzed/camera.pyx because it changed.
[1/1] Cythonizing pyzed/camera.pyx
running build
running build_py
running build_ext
building 'pyzed.defines' extension
error: Unable to find vcvarsall.bat

closed 2018-02-22 by Bauernteams

viniciusarruda, 2018-04-18:
How did you solved it ?

yifan-huang, 2019-04-09:
Have you got the solution to this problem yet? I got the same issue

finfrateam, 2020-02-29:
me,too.