stereolabs/zed-python-api #31: Set Mat Type

dagelabi, 2018-02-26:
Hi,

I'm trying to get a 32 bit depth map in python. To do so, when I initialize my Mat() I set those parameters :
```
mat = core.PyMat(	
			width = cam.get_resolution().width,
			height = cam.get_resolution().height,
			mat_type = core.PyMAT_TYPE.PyMAT_TYPE_32F_C1,
			memory_type = core.PyMEM.PyMEM_CPU
		)
```

However, when I run :
`print(mat.get_data_type())`

I still get `PyMAT_TYPE.PyMAT_TYPE_8U_C4`

Any ideas why ?

dagelabi, 2018-03-16:
**Update**

When I do this

	
```
mat = core.PyMat()

mat.init_mat_type(
			width = 1920,
			height = 1080,
			mat_type=core.PyMAT_TYPE.PyMAT_TYPE_32F_C1
		)
print(mat.get_data_type())
```

The console prints 
`PyMAT_TYPE.PyMAT_TYPE_32F_C1`

However, when I get the data and check out the type 

```
frame = self.mat.get_data()
print(frame.dtype)
```

I get 
`uint8`

Is it not supposed to be `float32` ?

A7ocin, 2019-03-05:
I have the same issue

closed 2022-04-27 by github-actions[bot]