stereolabs/zed-python-api #32: Swallowing up memory

Jam-zed, 2018-03-01:
I've noticed some memory issues with my ZED. I'm doing a project where I need a camera to run continuously for days or even more than that. But when I run the program it crashes after a 45 minutes. I've discovered that the problem have something to do with the "zed.grab()" function.

I've even tried using your Tutorial: Image capture to test if the same issue happens and it does. If i change the code so it run continuously instead of just taking 50 images then it will continue to eat up memory.

I'm watching the memory by using the linux top command.

Is there a method to clear the memory grab() is using?

Any help is more than welcome!

adujardin, 2018-03-01:
Hi,
This is indeed an issue. Nothing should be allocated _each_ grab(), only the first time, so there is no function to clear the memory.
Is this specific to the python wrapper or did you also notice this issue with the C++ SDK?
Which ZED SDK version are you using?
Are you on Tegra? (This shouldn't have any impact, it's just to help investigate the bug.)

Jam-zed, 2018-03-01:
I have only tried the python wrapper.

And I'm using ZED SDK 2.3.0

Right now I'm testing on my laptop a Dell xps 15 9560 running ubuntu 16.04, but I have also tested it on Tegra (astro carrier jetson tx1) and I'm having the same issue.

adujardin, 2018-03-01:
I think I found the issue.
I noticed removing the grab parameter declaration fix the (obvious) memory leak.
Basically replacing [this](https://github.com/stereolabs/zed-python/blob/master/tutorials/image_capture.py#L46) :

```python
    while i < 50:
        if zed.grab(zcam.PyRuntimeParameters()) == tp.PyERROR_CODE.PySUCCESS:
```

by this :

```python
    grab_params = zcam.PyRuntimeParameters()
    while i < 50:
        if zed.grab(grab_params) == tp.PyERROR_CODE.PySUCCESS:
```

Jam-zed, 2018-03-01:
That definitely made it a lot more stable!

Thank you very much!

commit 1674098e (2018-03-01): Fix #32 Leak memory from runtime params tutorials