stereolabs/zed-python-api #33: Inconsistent frame grabbing performance

gurus848, 2018-04-07:
Hi,
I am using the zed with a jetson TX2, with jetpack 3.2, zed sdk 2.3.3, python 3.5. I have noticed that when using the zed python library, sometimes the camera lags very badly, and the grab() function may take as long as 300 ms to run. Usually from my measurements it takes around 30 ms for the grab function to run, but the time fluctuates a lot and is usually above that. I am using 720p resolution, and performance depth mode. I am trying to track an object in real time with the camera, so this lag from the camera is causing major problems for me. Is there any way to increase the performance of camera and make it more consistent?

Thanks

gurus848, 2018-04-08:
I should mention that I am using a low exposure setting and a high gain setting to reduce motion blur in the image that is captured.

adujardin, 2018-09-13:
Hi,
In recent version of the ZED SDK, the grab time and the fluctuations were significantly improved, especially when using positional tracking on the TX2.
Also see #16

closed 2018-09-13 by adujardin