stereolabs/zed-python-api #35: Could I share my CUDA context with a PyZEDCamera?

oha08, 2018-04-09:
Though the ZED SDK provides a way to share my CUDA context with a ZED camera,
it seems that the current version of its python wrapper lacks it:

- InitParameters::sdk_cuda_ctx
- Camera::getCUDAContext

Is there any way to share my context with a PyZEDCamera?
I'm grad if ZED cameras could be used in CUDA applications developed on python.

adujardin, 2018-04-09:
Hi,
Yes, it's a missing feature, you currently can't (without modifying the plugin). 
It is planned though. It will probably require PyCUDA.

oha08, 2018-04-10:
Hi,
thank you for clarification. For now, I will try to modify the plugin to run it with PyCUDA.

oha08, 2018-06-01:
Hi,
I've noticed that I can use CUDA context switching (push & pop) to run my CUDA programs with PyZED rather than trying to share the contexts, which is sufficient for my purpose.
Yet it'd be convenient if context sharing will be implemented on PyZED in future :)

frk2, 2018-07-01:
@oha08 I'm new to this and trying to figure out how to do the same (I need to run a DNN and query the ZED In the same process). Any code you can share? How exactly does one switch cuda contexts? Thanks!

frk2, 2018-07-01:
I made a tiny cython module to do this and it works! Thanks for the tip guys! 

https://github.com/frk2/cuda-fun

oha08, 2018-07-02:
@frk2 Hi, Your module looks pretty nice and simple enough for working! In my case, I'm using PyCUDA to create and push/pop my context. Because PyCUDA doesn't provide any interface to get the current context, I've selected to keep the context made by ZED stacked and push my context on top of it:

1) open ZED Camera where a context is created
2) create my context, which is automatically stacked on top of the previous context
3) pop my context when calling ZED functions
4) push my context when calling my CUDA kernels (if it is popped before)
and so on

Your module enables more flexible control than this :)  Thanks a lot!

frk2, 2018-07-02:
I was thinking of doing the same - but the bummer is that you can't pass a created context to either ZED or Pytorch (they both create one) and hence pycuda was useless since it doesn't let me get the current context.

Thanks for the tip though definitely guided me in the right direction!

adujardin, 2018-07-04:
@frk2 By default the ZED SDK creates a new context, but with the C++ SDK you can pass your own context to the ZED with [InitParameters::sdk_cuda_ctx](https://www.stereolabs.com/developers/documentation/API/v2.4.0/structsl_1_1InitParameters.html#ae684a455a5dc2e48a82b9aadc8c3ba53).
As mentioned in the original post, this feature is missing from the python wrapper though.

yuvalg1987, 2021-10-03:
Any update whether context sharing is possible with ZED camera and the python API? I'm trying to work with the ZED camera an CuPy an have probably encountered an issue related to this.

yuvalg1987, 2021-10-04:
Just to update on my solution for this issue. The ZED runs its own context internally and therefore processing images using CuPy should be handled in a different thread. My best solution relied on the ZED-TensorFlow example which worked pretty good.

zcy1113, 2023-03-11:
你好。我用的是zed相机，cuda是11.7，  运行get_python_api.py文件时，提示PermissionError: [Errno 13] Permission denied: 'C:\\Program Files (x86)\\ZED SDK\\pyzed-3.8-cp39-cp39-win_amd64.whl'，但是安装whl文件之后，继续重复出现这个错误，是什么原因导致的呢

zcy1113, 2023-03-11:
Hello. I use a zed camera, and the cuda is 11.7. When I run the get_python_api.py file, I am prompted with permission error: [errno13] Permission denied:' C: \ \ Program files (x86) \ \ ZED SDK \ \ PYZED-3.8-CP39-CP39-Win-amd64.whl', but after installing the whl file, what causes this error to appear repeatedly?