stereolabs/zed-python-api #36: Pre determine SVO length

ajdroid, 2018-04-23:
Hi, I'm trying to follow the read svo example to stream and process an svo I recorded. I want to store all of the depth frames (say in a numpy array). Is there a way to determine the length of an svo without streaming all the way to the end?

obraun-sl, 2018-04-24:
Hi,
You can use Camera.getSVONumberOfFrames() to know the number of frames stored in a SVO. (Open() must be called before).

ajdroid, 2018-04-24:
I have a slightly unrelated question as well and I'd be grateful if you could answer it here. I'm not sure how to interpret the ZED depth data except in relative terms (eg. pixel A is twice as far from the camera as pixel B; rather than pixel A is 10 cms away and pixel B is 5cms away)

It seems like this is doable since we have two cameras and we know the transformation between them and their extrinsic and intrinsic parameters. 

So, my question is, does the depth data have a conversion to a metric 'unit' of distance from the camera (mm/cm) or does the zed API provide a way to triangulate the 3D points from the depth data?

obraun-sl, 2018-04-24:
Hi,

Yes, the ZED SDK provide the depth in metric distance with `camera.retrieveMeasure(MEASURE::DEPTH) `function. The unit will be in mm by default but you can change them with the InitParameters.coordinate_units parameter. For each pixels, the measure is an absolute position to the camera and not relative between pixels.

If you want to have the 3D point of the pixel, you can also use `camera.retrieveMeasure(MEASURE::XYZ) `.
Each pixel will contain a float4 value with (x,y,z) the coordinate of the 3D point defined by the pixel.

ajdroid, 2018-04-24:
Great! Why is a 4D vector returned? Are these homogeneous coordinates?

obraun-sl, 2018-04-24:
Because you also have the XYZRGBA (point cloud with color) where the last float contains the RGBA value (4uchar inside a float). To have it more consistent with the other parameters, it's a float4 even if the last float is not used in XYZ

ajdroid, 2018-04-24:
Gotcha. I see that the last parameter is 0 when I use `PyMEASURE.PyMEASURE_XYZ`