stereolabs/zed-python-api #41: Reduced field of view when using retrieve_image

F0lha, 2018-06-18:
I found that the field of view of the camera is reduced when using retrieve_image when comparing with directly using the ZED SDK application. This behavior is also found on the example of [live_camera.py](https://github.com/stereolabs/zed-python/blob/master/examples/live_camera.py).

However, when using OpenCV's VideoCapture I can obtain the image with the full field of view.

Can you help me solve this problem?

adujardin, 2018-06-18:
Hi
The images are cropped by the stereo rectification to align the images, it is not an issue. This operation is needed to compute the distances.

You can retrieve the unrectified images with [`VIEW_LEFT_UNRECTIFIED`](https://www.stereolabs.com/developers/documentation/API/v2.4.0/group__Video__group.html#ga77fc7bfc159040a1e2ffb074a8ad248c)
But keep in mind that the depth map won't be aligned with these images.

closed 2018-06-18 by adujardin

F0lha, 2018-06-18:
Thanks for the help!