stereolabs/zed-python-api #47: Using the ZED SDK without an svo file

ajdroid, 2018-07-23:
@obraun-sl Hi Oliver!

 Is it possible to pass the right and left images through the ZED SDK as if simulating reading an svo?

I have a situation where I do not have access to the ZED camera itself and some data was recorded on a ZED camera (with known calibration). Left and right image pairs themselves were stored but not in an svo.

ajdroid, 2018-07-26:
The answer is no!

closed 2018-07-26 by ajdroid

mminervini, 2019-07-30:
Hi,

Sorry for resuming this closed issue...
@ajdroid's answer is crystal clear, but I'm working with multiple ZED cameras on the same computer and I was wondering if anything has changed in the last year on this front, i.e. bringing back to the ZED SDK a video that was acquired with a ZED camera and exported to a format other than SVO.

My application runs on an embedded platform shipping an Nvidia GeForce card that supports up to 2 concurrent NVENC sessions (like all GeForce), but I need to record from 3 ZED cameras at the same time (the three videos should be as synchronized as possible) and as far as I understood SVO recording in AVCHD or HEVC modes can only operate on NVENC.
To circumvent the limitation, I was thinking about recording two streams in SVO and the third in some other format, but I'm realizing that once a video leaves the ZED SDK proprietary format apparently it's not possible to bring it back to the SDK to actually use it (e.g., compute rectified images, depth, etc.).

Is there a way around this issue that lets me record more streams than allowed by the NVENC limit while safely remaining within the SDK?
For instance, is there (or there will be) a video (H.264 or HEVC) recording mode that relies on the CPU? (Recording in PNG or JPG would result in huge disk occupancy and for hours of videos does not represent a viable option.)

Thanks in advance for the support!

Best,
Massimo

adujardin, 2019-07-30:
Hi @mminervini
Unfortunately, at this time the only available options are :
1. Choosing a different compression mode for the SVO recording
2. Using a [Tegra or Quadro cards](https://developer.nvidia.com/video-encode-decode-gpu-support-matrix) which don't have NVENC limitation. The cheapest option would probably be either the Quadro K2000 for desktop or maybe the Jetson Nano (with lower framerates and limited compute power) or the Xavier which support extension card to handle the USB bandwidth.

We plan to add other encoding support alternatives in the future (such as Intel Quick Sync), but we can't give you the timeline yet.

Technically, but that's not a clean workaround, I guess you could record the SVO in image-based compression split into chunks, then convert them in the background in H26X using NVENC (with SVO Editor called from a script) for archiving smaller files.