stereolabs/zed-python-api #48: Use multiple ZED camera

ralf307, 2018-08-03:
Hello,

thanks a lot for your sharing! However, here I still have a quetion:

when my computer is connected with multiple ZED camera (e.g., two), then how can I access them separately using some identifier as parameter in your code? I just want to use those two ZED cams to take pictures in different directions. 

Best regards,
XG

ultrafro, 2018-09-05:
Thanks for sharing! This is a great extension of the SDK. I think the multiple ZED is an important feature and it would be great to see it in the python SDK. Is there any plan to include this? If not, any tips how we can extend it?

Thansk!

adujardin, 2018-09-13:
Hi, 
This is already implemented, you have to set the camera ID 
[`PyInitParameters.camera_linux_id`](https://github.com/stereolabs/zed-python/blob/master/pyzed/camera.pyx#L107)

Here is a simple example with 2 ZED: https://gist.github.com/adujardin/14843cb6f6f1380ad5ca9b948f85c531

closed 2018-09-13 by adujardin

nalinraut, 2019-01-28:
import cv2
import pyzed.sl as sl
zed1 = sl.Camera()
zed2 = sl.Camera()

def main():
	print("Running...")
	init_params = sl.InitParameters()
	init_params.camera_resolution = sl.RESOLUTION.RESOLUTION_HD720
	init_params.camera_fps = 30
	
	
	init_params.camera_linux_id = 0
	if not zed1.is_opened():
		print("Opening ZED Camera 1 ...")
	
	err = zed1.open(init_params)
	if err != sl.ERROR_CODE.SUCCESS:
		print(repr(err))
		exit(1)
	

	init_params.camera_linux_id = 1 # Specify the camera index
	if not zed2.is_opened():
		print("Opening ZED Camera 2 ...")
	
	err = zed2.open(init_params)
	if err != sl.ERROR_CODE.SUCCESS:
		print(repr(err))
		exit(1)

	runtime = sl.RuntimeParameters()
	mat1 = sl.Mat()
	mat2 = sl.Mat()

	print_camera_information(zed1)
	print_camera_information(zed2)

	key = ''
	while key != 113:  # for 'q' key
		err = zed1.grab(runtime)
		if err == sl.ERROR_CODE.SUCCESS:
			zed1.retrieve_image(mat1, sl.VIEW.VIEW_LEFT)
			cv2.imshow("ZED 1", mat1.get_data())

		err = zed2.grab(runtime)
		if err == sl.ERROR_CODE.SUCCESS:
			zed2.retrieve_image(mat2, sl.VIEW.VIEW_LEFT)
			cv2.imshow("ZED 2", mat2.get_data())

		key = cv2.waitKey(5)
	cv2.destroyAllWindows()

	zed1.close()
	zed2.close()
	print("\nFINISH")

def print_camera_information(cam):
	print("Resolution: {0}, {1}.".format(
		round(cam.get_resolution().width, 2), cam.get_resolution().height))
	print("Camera FPS: {0}.".format(cam.get_camera_fps()))
	print("Firmware: {0}.".format(
		cam.get_camera_information().firmware_version))
	print("Serial number: {0}.\n".format(
		cam.get_camera_information().serial_number))

if __name__ == "__main__":
	main()

mminervini, 2019-07-25:
Hi,

I am trying to select a camera in a multi-camera setting based on the serial number, using the `set_from_serial_number()` function from `InitParameters.input`, however, when I `open()` with those `init_params`, the serial number appears to be ignored and another camera is opened.
Even if I use a random non-existent serial number, upon opening no error is issued and the same _default_ camera is accessed.

Is `set_from_serial_number()` reliable or `camera_linux_id` is the only viable option to select a camera? (I'm in a Linux environment.)

Thanks in advance!

Best,
Massimo

Poutsh, 2019-07-25:
Hi @mminervini ,

To fix your problem you can use the function ```set_from_serial_number()``` from InitParameters so use :
```
init.set_from_serial_number(num)
```
instead of : 
```
init.input.set_from_serial_number(num)
```
You can also create an object ```InputType``` use the function and then assign it to the ```InitParameters``` but this is more laborious. We will soon remove this ambiguity to avoid confusion so you can just use the set_from functions directly for ```InitParameters```.

Thanks for your report.