stereolabs/zed-python-api #49: Ability to set custom range

rukie, 2018-08-13:
Using the commands PyRANGE_FAR, MEDIUM, and NEAR result in distances of 2, 1, and 0. I would like to be able to pass in a value. How is this done?

> spatial.set_range(range=zcam.PyRANGE.PyRANGE_FAR)

rukie, 2018-08-13:
Also: get_range always returns the same results, no matter whats provided in the set_range command.

adujardin, 2018-09-13:
Hi,
These functions are a bit confusing. The issue come from the C++ API which has static functions to get the values from the enum and other to actually set the values (see [C++ API Doc](https://www.stereolabs.com/developers/documentation/API/v2.6.0/structsl_1_1SpatialMappingParameters.html)).

To set/get the actual value you have to do something like this :

```python
import pyzed.camera as zcam

spatial = zcam.PySpatialMappingParameters()
spatial.range_meter = 2 # Manually set the range
spatial.range_meter = spatial.get_range(zcam.PyRANGE.PyRANGE_NEAR) # Set the range using the recommended template
range_value = spatial.range_meter # Get the value
print(range_value)
```
I'll try to fix this part with a clearer solution in python wrapper.

adujardin, 2018-09-13:
Update: I renamed the static get functions :

`get_range` is now `get_range_preset`
`get_resolution` is now `get_resolution_preset`

I removed the default value which didn't make sense.

commit ac8b99a4 (2018-09-13): Fix #49 Renamed get range and resolution from SpatialMappingParams to get_..._preset()
closed 2018-09-13 by adujardin