stereolabs/zed-python-api #55: ImportError: DLL load failed: The specified module could not be found.

JonesParking18, 2018-10-24:
I could run setup.py without any problem. I can even import pyzed. But I cannot import pyzed.camera (nor other pyzed.* modules) because of 
`ImportError: DLL load failed: The specified module could not be found.`
error. I looks like it is a problem with a path. Does anyone know how to fix this?
I am running this on Windows10 with Anaconda, python 3.5.6. Let me know if you need more information.

closed 2018-11-09 by JonesParking18

linthy, 2019-01-19:
Hello, do you solve this issues?
I meet the same problem today.
I am running this on Windows10 with Anaconda, python 3.5.5.

hmc-cs-syasunaga, 2019-01-19:
Hi. It seems like the problem was with the installation of the CUDA toolkit. 
I deleted all of the installed CUDA-related softwares and installed it again and it worked!