stereolabs/zed-python-api #56: 24 is not a valid PyERROR_CODE

Parsifal619, 2018-11-03:
when i try to open zed camera, the error heppens:
but i think the error is not from my code, because zed ran perfect on my compter before
`from ._conv import register_converters as _register_converters
Using TensorFlow backend.
model_data/trained_weights_final.h5 model, anchors, and classes loaded.
Traceback (most recent call last):
  File "new2.py", line 328, in <module>
    detect_video(yolo, video_path, output_path="model_data/detected_cone4.mp4")
  File "new2.py", line 208, in detect_video
    err=zed.open(init_params)
  File "pyzed\camera.pyx", line 583, in pyzed.camera.PyZEDCamera.open
  File "C:\ProgramData\Anaconda3\lib\enum.py", line 291, in __call__
    return cls.__new__(cls, value)
  File "C:\ProgramData\Anaconda3\lib\enum.py", line 533, in __new__
    return cls._missing_(value)
  File "C:\ProgramData\Anaconda3\lib\enum.py", line 546, in _missing_
    raise ValueError("%r is not a valid %s" % (value, cls.__name__))
ValueError: 24 is not a valid PyERROR_CODE`

Parsifal619, 2018-11-03:
Can someone help me solve this? Thanks.

adujardin, 2018-11-05:
Hi,
If you are using the latest python wrapper, it shouldn't happen, as 24 is a valid PyERROR_CODE (Camera already is use) :

https://github.com/stereolabs/zed-python/blob/0c8ba468c1927a01028dd35b4c53134209b3af4c/pyzed/types.pyx#L55

If the value would have been bigger it could have been because the python wrapper is currently a little behind the C++ SDK. So there are some missing features, including missing ERROR_CODE. The [C++ documentation](https://www.stereolabs.com/docs/api/group__Camera__group.html#ga4db9ee29f2ff83c71567c12f6bfbf28c) contains the correct ERROR_CODE list.

closed 2022-04-27 by github-actions[bot]