stereolabs/zed-python-api #59: Running ZED camera with CUDA 9.1 and ZED SDK 2.4, but got some error message

vb123er951, 2018-11-27:
The README said this is for ZED SDK 2.3, can it run with SDK 2.4?
My computer is using CUDA 9.1 and Cudnn 7.1.3, and I installed ZED SDK 2.4 version, otherwise the camera can't work.

When I running the program in tutorial directory, it output this message:
```
ZED (Init) >> ERROR: Executable was designed for SDK v2.3.3 / installed v2.4.1

ZED (Init) >> Video mode: HD720@60
```

vb123er951, 2018-11-28:
Hi, I tried to re-compile/ re-install the packages, I use the commands below:
```
python3 setup.py build
python3 setup.py install
```
it doesn't have error messages, but after install I used the command
`python3 -c "import pyzed.camera as zcam`
it shows ImportError: no module named 'pyzed.camera'

Did I miss something?

adujardin, 2018-11-29:
Did you install it with enough permission?
e.g. `sudo python3 setup.py install`

vb123er951, 2018-11-29:
Yes, I did.

vb123er951, 2018-11-29:
I have checked the python packages, the packages have been installed.
And I tried to import command again, it imported successfully, but I don't know why suddenly works well.
Anyway, thank you for the assistance!

closed 2018-11-29 by adujardin