stereolabs/zed-python-api #65: my python can not find the pyzed.s1 model.

k98732142, 2018-12-24:
I use anaconda3 to create the environment "py36" and i also success to use 'python setup.py build' and 'python setup.py install'.
But my python can not find the pyzed.s1 model.
[image]

Poutsh, 2019-01-02:
Hi @k98732142 

I think you have made a typo here. It's not pyzed.s1 but pyzed.sl, but you seem to have wrote that in your code on your screenshot.
Have you installed the module using pip like said in the README? If you are using Anaconda, it'll be in a totally different path and will not work, you should try without it.

k98732142, 2019-01-04:
@Poutsh  Yes....I write the wrong model name. Thanks you for your help.

closed 2019-01-04 by k98732142