stereolabs/zed-python-api #66: No module named 'pyzed.camera'

Claire06, 2019-01-09:
Hello, thank you for this great project.
I followed all the Stereolabs ZED - Python Integration (beta) github's instructions but I got some issues importing pyzed.camera, pyzed.core, pyzed.define, pyzed.types, pyzed.mesh afterwards.

I've tried with different version of python (3.5, 3.6, 3.6) and I am actually using Cuda 10.0.130.
Do you have an idea of what the problem could be ?
We suspect the pyzed compilation is failling silently but are not sure. Any help will be appreciated, thanks again.

Poutsh, 2019-01-11:
Hi @Claire06 ,

If you've cloned the last version of the zed python api on this github repository, you shouldn't use the old imports camera, core, types etc. After installing the module, you should simply use :

```
import pyzed.sl as sl
```

Please tell me if that fixed your problem.

Claire06, 2019-01-11:
Hi @Poutsh ,
Thank you very much for your reply.
I am actually trying to use the imports camera, core, types etc. because I am trying to run the object_detection_zed.py file.
I followed all the 3D Object Detection using ZED and Tensorflow github's instructions, and I thought that all the imports would have been accessible after building and installing the setup.py.

So my question is how could I install those modules (pyzed.camera, pyzed.core, pyzed.define, pyzed.types, pyzed.mesh) ?

Thank you again for helping me, 
Claire

Poutsh, 2019-01-14:
Hi @Claire06 ,

The 3D Object Detection Project for ZED has also been updated since then according to the new import pyzed.sl and no more uses the old imports (pyzed.camera etc).

As you can see it : https://github.com/stereolabs/zed-tensorflow/blob/master/object_detection_zed.py

You should use this version of the project instead.

Claire06, 2019-01-14:
Hello @Poutsh,
Thank you very much for your reply, it helped me a lot !

closed 2019-01-14 by Poutsh