stereolabs/zed-python-api #70: Spatial Mapping with ZEDm using the Pyhton API

nalinraut, 2019-01-28:
I tried the spatial mapping example using the Python API, however, there seems something wrong with code since it exits at the enabling stage. For more details have a look below at the snippets.

[image]

[image]

Poutsh, 2019-01-29:
Hi @nalinraut 

Can I ask you what is the ZED SDK version you're using and on which platform you're running that program?

nalinraut, 2019-01-29:
@Poutsh I am using ZED SDK 2.7 on TX2. Using python3. I get 'FAILURE' as error when i print it.

nalinraut, 2019-01-29:
`import pyzed.sl as sl

def main():
    zed = sl.Camera()
    init_params = sl.InitParameters()
    init_params.sdk_verbose = True
    err = zed.open(init_params)
    print(repr(err))
    zed.close()

if __name__ == "__main__":
    main()`

KevinGriffin31, 2019-04-01:
Also having this same issue has anything new been found out about it?

KevinGriffin31, 2019-04-01:
@adujardin after running the code you posted I get a similar output as @nalinraut got, however, I am missing a line. 
I got:

ZED (Init) >> Depth mode: PERFORMANCE
ZED (Init) >> Video mode: HD720@60
SUCCESS

adujardin, 2019-04-01:
You can display the error code information to know more, something like this: 
```python
err = zed.enable_spatial_mapping(mapping_parameters)
if err != sl.ERROR_CODE.success:
	print(repr(err))
```

KevinGriffin31, 2019-04-08:
The above code only returned "FAILURE"

closed 2022-04-27 by github-actions[bot]