stereolabs/zed-python-api #72: Depth image

zaher88abd, 2019-02-26:
Hi,

I would ask about depth image, why I am getting an image with four channels first three are the same, and the last one is a white image?

Poutsh, 2019-04-01:
Hi @zaher88abd ,

You are probably retrieving the depth image using retireveImage and not using retrieveMeasure. You should retrieve the depth like this, and it will give you a 32FC1 matrix.

```
zed.retrieve_measure(depth, sl.MEASURE.MEASURE_DEPTH)
```

You can check the depth sensing tutorial here : [tutorials/depth_sensing.py](https://github.com/stereolabs/zed-python-api/blob/master/tutorials/depth_sensing.py)

zaher88abd, 2019-04-30:
Thanks, @Poutsh. I got the data but when I tried to show it by opencv I got some issues. Firstly, the coding of the image is float32. SO, what is the best way to show it to uint8? Or there is a diffrenet way without casting.

zaher88abd, 2019-04-30:
Thanks, @Poutsh.

closed 2019-04-30 by zaher88abd