stereolabs/zed-python-api #78: cannot import pyzed.sl

Lakshaysharma048, 2019-04-30:
Hi,
sdk 2.8
cuda 10.0
python 3.5

i went for python wrapper and did
sudo python3 setup.py build
sudo python3 setup.py install

there are no errors, but when i try to do:

import pyzed.sl as sl, it says: no module named pyzed.sl

Because of this darknet.py is also not running.

My issue is similar to old issue but for him it surprisingly imported after sometime.

I event rebooted the system.

i can see it pyzed gets installed in root path...usr/local/...

but it's not getting imported. Is it because my python is installed from conda?

samgithub11, 2019-05-05:
After successfully installing SDK and its dependencies, 
I am also facing the same issue when I run any example or tutorial code. It gives the error that "No module pyzed.sl". 
I am using Pycharm and Anaconda.
It shows that pyzed is installed in 
Removing C:/Users/Sammm/Anaconda3/Lib/site_packages/pyzed-2.8-py3.7.egg-info
Writing C:/Users/Sammm/Anaconda3/Lib/site_packages/pyzed-2.8-py3.7.egg-info

But 

**ERROR 1. **** When I run  live_camera.py  example or any other example or tutorial script, it gives error:  No module named pyzed.sl as shown below****** 

 
Traceback (most recent call last):
  File "C:/Users/sammm/PycharmProjects/zed-python-api-master/examples/live_camera.py", line 27, in <module>
    import pyzed.sl as sl
ModuleNotFoundError: No module named 'pyzed.sl'
 

**ERROR 2.  **** When I run following script for camera configuration, it gives error: module  pyzed  has no attribute  Camera****** 

SCRIPT:
 import pyzed as sl
zed = sl.Camera() #Create   ZED camera object
init_params = sl.InitParameters() # Set configuration parameters
init_params.camera_resolution = sl.RESOLUTION.RESOLUTION_HD1080
init_params.camera_fps = 30
err = zed.open(init_params) # Open the camera
if err != sl.ERROR_CODE.SUCCESS:
    exit(-1) 
 
I am using
SDK: 2.8
Python 3.5, 3.6, 3.7 (I tried different python versions)
cuda 10
OS: Windows 10

So,
Need help to know what is the issue?

CamiloPardoBeainy, 2019-05-06:
hi, 
I have a similar problem

I do not have errors in the installation
of the API, but when I run an example I have this error:

---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
<ipython-input-1-579c251116f7> in <module>
     19 ########################################################################
     20 
---> 21 import pyzed.sl as sl
     22 import math
     23 import numpy as np

ModuleNotFoundError: No module named 'pyzed.sl'

---------------------------------------------------------------------------

I appreciate if you can please help me

hankhhub, 2019-05-16:
I used conda for python as well. It will work only if you change directory to pyzed.

orcund, 2019-05-16:
> I used conda for python as well. It will work only if you change directory to pyzed.

> hi,
> I have a similar problem
> 
> I do not have errors in the installation
> of the API, but when I run an example I have this error:
> 
> ModuleNotFoundError Traceback (most recent call last)
> in 
> 19 ########################################################################
> 20
> ---> 21 import pyzed.sl as sl
> 22 import math
> 23 import numpy as np
> 
> ModuleNotFoundError: No module named 'pyzed.sl'
> 
> I appreciate if you can please help me

Hey, had the same problem. In the directory that you installed Anaconda go to "envs\[YourEnvName]\Lib\site-packages\pyzed", there is a Python Extension File starting like "sl.cp" copy it and paste to "pyzed" dir of the "zed-python-api-master" you downloaded from git and used to install. That solved my problem. Hope it works for you as well.

Wim-DEV, 2019-06-30:
Hi,

I have installed using pip and have the similar error. I have tried to install building the package from the git download and it throws another error regarding ` fatal error: sl/Camera.hpp: No such file or directory`. This is my current system setup:

Python 3.6.8

Cuda 10

SDK 2.8.1.

Ubuntu 18.

Further as I can only achieve a pip installation copying things to the api base repo has no effect for my project. Further on run the package does not complain about the import but rather the sdk verion:

`ZED (Init) >> ERROR: Executable was designed for SDK v2.8.0 / installed v2.8.1`

Please note this occurs even though my IDE cannot find the module `pyzed.sl`. So I seem to have 3 separate issues I cannot solve. 

I have looked for legacy version of the sdk and will looks again. Any ideas besides re-installing a legacy version of the sdk?

asmaaelkeurti, 2019-09-01:
> Hey, had the same problem. In the directory that you installed Anaconda go to "envs[YourEnvName]\Lib\site-packages\pyzed", there is a Python Extension File starting like "sl.cp" copy it and paste to [...]

"there is a Python Extension File starting like sl.cp"
i m sorry, r u sure it's an python extension file ? the only file started by sl.cp is .so file

noelquah, 2020-07-01:
Hi,
I followed your instructions and installed ZED3.2/CUDA 10.2/Cython/Numpy and followed the install script procedures.
But i am encountering some problems and it would be great if someone could advise.

When i run python with pyzed.sl, there is an error message:

> Traceback (most recent call last):
>   File "<file directory>", line 1, in <module>
>     import pyzed.sl as sl
> ImportError: DLL load failed while importing sl: The specified module could not be found.

upon checking my pyzed installation via pip freeze, i get this after using the command 
(python -m pip install pyzed-3.2-cp38-cp38-win_amd64.whl):

> pyzed @ file:///C:/Program%20Files%20%28x86%29/ZED%20SDK/pyzed-3.2-cp38-cp38-win_amd64.whl 

i am aware that i am supposed to get this:
> pyzed==3.2

It seems like an installation issue, how can i resolve this matter?
Thanks in advance (im throwing in some images in hopes that it may help visually)!

[image]
[image]

BryanBo-Cao, 2021-01-07:
Tried installing [SDK](https://www.stereolabs.com/developers/release/) first?

kunalkumthekar, 2021-02-17:
Hello everyone,
im trying to install the pyzed api with no luck. python version 3.9, latest version of sdk and cuda. Not able to install manually as I get the error "Unable to find vcvarsall.bat". Also there is an import error for pyzed.sl . Any help would be appreciated.
Thanks

acherner, 2021-03-24:
same issue : pyzed.sl not recognized
Win 10 64 , Intellij, VS
installed Anaconda reconfigured the virtual env for ZED, reinstalled the SDK - same result
played with moving the sl.cp file no success, the file itself appears to be ok:
copying build\lib.win-amd64-3.9\pyzed\sl.cp39-win_amd64.pyd -> C:\Users\PC\AppData\Local\Programs\Python\Python39\Lib\site-packages\pyzed

qt-truong, 2021-03-24:
Hi,

If you are using a virtual environment, make sure to install the Python API **inside** the environment itself. Here are instructions on how to do so on Windows : [virtual env doc](https://www.stereolabs.com/docs/app-development/python/virtual_env/)

Also, on Windows, there can be issues with Python versions >= 3.8. If it is possible for you, could you try with a former version of Python ?

@kunalkumthekar your issue seems to be related to your C++ compiler. Could you try the suggested solution mentioned in https://github.com/stereolabs/zed-python-api/issues/147#issuecomment-780539266 ?

acherner, 2021-03-25:
> Hi,
>
> If you are using a virtual environment, make sure to install the Python API **inside** the environment itself. Here are instructions on how to do so on Windows : [virtual env [...]

that is exactly the issue, I had several Python versions , including 3.9 and had my IDE misconfigured.
So the best way to solve it on Windows is using Anaconda - configure the virtual environment for ZED, run it's console , install everything starting from dependencies and choose the virtual env as project SDK in IDE.
So at this point i'm running SDK on 3.8 without any problems.

kunalkumthekar, 2021-03-25:
> same issue : pyzed.sl not recognized
> Win 10 64 , Intellij, VS
> installed Anaconda reconfigured the virtual env for ZED, reinstalled the SDK - same result
> played with moving the sl.cp file no success, [...]

When I installed the library I had issues with python version > = 3.7 . Try and install the library within the conda env with a python version of 3.6 installed in it.

closed 2022-04-27 by github-actions[bot]