stereolabs/zed-python-api #79: Convert pixel width to mm

zaher88abd, 2019-05-02:
I want to convert the width of an object in the image from pixels to cm: depends on this [link ](https://computergraphics.stackexchange.com/questions/3752/estimating-real-world-object-distance-from-photo) I need to calculate this equation:
object_image_size = (real_object_size * lens_focal_length)/object distance
From your website, the cameras have sensor 1/3" which is 4.8X3.6MM, but you show the focal length in pixels is that mean for HD720 the focal length is (700*0.004 ) = 2.8 MM. is that right?

Machoney, 2019-05-16:
I'm joining the question. It is also important for me. Or do you recommed to convert depth to PCL and then check the object dimensions?

zaher88abd, 2019-07-29:
Please, can we get an answer to this question?

adujardin, 2019-07-29:
I may not have understood the question but since the depth information is available, and you have 2 "bounds" pixel. You can get the 3D points coordinates for each and compute the euclidian distance. That will give you the object size in metrics from 2 pixels.

Here is a snippet of what I mean :

```python
import pyzed.sl as sl
import math
import numpy as np

def main():
    zed = sl.Camera()

    init_params = sl.InitParameters()
    init_params.coordinate_units = sl.UNIT.UNIT_METER

    err = zed.open(init_params)
    if err != sl.ERROR_CODE.SUCCESS:
        exit(1)

    runtime_parameters = sl.RuntimeParameters()
    point_cloud = sl.Mat()

    zed.grab(runtime_parameters)
    zed.retrieve_measure(point_cloud, sl.MEASURE.MEASURE_XYZRGBA)

    # Object 2D bounds
    b1_u = 110
    b1_v = 200

    b2_u = 150
    b2_v = 200

    # Get the 3D coordinates for these positions
    err, b1_3d_value = point_cloud.get_value(b1_u, b1_v)
    err, b2_3d_value = point_cloud.get_value(b2_u, b2_v)

    # Euclidian Distance between the two 3D points
    dist = math.sqrt((b1_3d_value[0] - b2_3d_value[0]) * (b1_3d_value[0] - b2_3d_value[0]) +
                            (b1_3d_value[1] - b2_3d_value[1]) * (b1_3d_value[1] - b2_3d_value[1]) +
                            (b1_3d_value[2] - b2_3d_value[2]) * (b1_3d_value[2] - b2_3d_value[2]))

    print("Object size ", dist)
    zed.close()

if __name__ == "__main__":
    main()
```

Would that work for your application?

zaher88abd, 2019-07-29:
Thanks, @adujardin for the reply. But I am working on estimating the depth using deep learning model and I am using the Zed camera depth view as the ground truth. So, I need to know the focal length in mm to calculate the depth from the output of the model. I hope, I clarify why I need the length in mm.

adujardin, 2019-07-29:
The focal in mm can be found by using [this formula](https://github.com/stereolabs/zed-opencv/issues/39#issuecomment-311611622) :
```
f_mm = (fx+fy) / 2 * pixel_size_mm
```

If you know exactly the size of the object and you are able to detect it, then you could have an estimation of the distance using the method described at the link.

However, if the deep learning estimation algorithm output the disparity then you can compute the 3D coordinates. The "metric" part is provided by the distance between the 2 camera sensors, also known as the baseline, not the focal. 

You can't compute absolute distances with only 1 camera (and no other knowledge), even with calibrations. 

The [depth is computed from the calibrated focal](https://docs.opencv.org/3.0-beta/doc/py_tutorials/py_calib3d/py_depthmap/py_depthmap.html) *in pixel*, the disparity between the left and right image and the transformation matrix between the 2 sensors.

```
depth = focal_x * baseline / disparity
x = ( (u - principal_point_x) * depth) / focal_x
y = ( (v - principal_point_y) * depth) / focal_y
```

If the algorithm outputs a pseudo depth, for instance using only the left image, then you can't get metrics values without estimating first the absolute scale (similar to the monocular odometry problem).

zaher88abd, 2019-07-29:
Thank you @adujardin, for explaining.

closed 2019-07-29 by zaher88abd