stereolabs/zed-python-api #82: Different FPS from the setting

shgold, 2019-05-31:
Hello, 
I am using ZED camera under linux 18.04 with CUDA 10 and plugged the device to USB3.0 ports.

I am trying to run ZED initialized with HD1080@30fps. However, when I print out `zed.get_current_fps()` for every frames grabbed, it prints out different values between 10 to 15. 

I tried to change the `camera_fps` parameter for the init still it did not work. 
I would like to know what could be the possible reason for having this issue.

Best regards,
Stella

Myzhar, 2019-05-31:
Dear @shgold 
the command `get_current_fps` returns the real FPS of the camera.
If you get 10-15FPS it means that your system is not powerful enough to support HD1080@30fps.

What is your system configuration? (CPU, GPU, ...)

shgold, 2019-06-04:
My computer's CPU is  Intel(R) Core(TM) i7-8550U CPU @ 1.80G<Hz 
and the GPU version is Nvidia Geforce 940MX

shgold, 2019-06-13:
@Myzhar 
I tried the same script on the different computer with *GeForce GTX 1050 Ti* and `zed.get_current_fps()` gives correct fps that I have set. 

Then I've recorded video with HD1080@30fps with ULTRA_DEPTH_MODE configuration. 
I've recorded video for about 20seconds however, the total number of frames in this SVO file is only 300 frame which result in 15fps~16fps. 

Why is the fps changes before and after recording? Is it because that I am using ULTRA_DEPTH_MODE?

closed 2022-04-27 by github-actions[bot]