stereolabs/zed-python-api #84: NOT A NEW FRAME

zaher88abd, 2019-06-11:
Hi, 

I am using Zed Camera with Python API, and SDK 2.8. On Ubuntu Machine 16.4 and it has Cuda 10.1. I am trying it by this code:
```import pyzed.sl as sl
import cv2
import numpy as np

def run_zed_camera():
    try:
        global cam
        print("Running...")
        init = sl.InitParameters()
        init.coordinate_units = sl.UNIT.UNIT_METER
        init.camera_resolution = sl.RESOLUTION.RESOLUTION_HD720
        # init.depth_mode = sl.DEPTH_MODE.DEPTH_MODE_NONE
        # init.depth_stabilization = False
        # init.depth_minimum_distance = 0.50
        init.svo_real_time_mode = False
        init.camera_fps = 30
        cam = sl.Camera()
        status = cam.open(init)
        if not cam.is_opened():
            print("Opening ZED Camera...")
        if status != sl.ERROR_CODE.SUCCESS:
            print(repr(status))
            exit()

        runtime = sl.RuntimeParameters()
        runtime.sensing_mode = sl.SENSING_MODE.SENSING_MODE_FILL

        print_camera_information(cam)
        # cam.set_depth_max_range_value(20)
        err = cam.grab(runtime)

        if err == sl.ERROR_CODE.SUCCESS:
            return cam, runtime
        else:
            print("Error the zed camera not working,", err)
            cam.close()
            exit()
    except Exception as e:
        cam.close()
        print("Error", e)

def get_frame(cam, runtime):
    try:
        mat = sl.Mat()
        err = cam.grab(runtime)
        if err == sl.ERROR_CODE.SUCCESS:
            cam.retrieve_image(mat, sl.VIEW.VIEW_LEFT, width=640, height=480)
            frame = load_image_into_numpy_array(mat)
        else:
            print("get Frame:", err)
            cam.close()
            exit()

        return frame
    except Exception as e:
        print("Error", e)
        cam.close()

def print_camera_information(cam):
    try:
        print("Resolution: {0}, {1}.".format(
            round(cam.get_resolution().width, 2), cam.get_resolution().height))
        print("Camera FPS: {0}.".format(cam.get_camera_fps()))
        print("Firmware: {0}.".format(
            cam.get_camera_information().firmware_version))
        print("Serial number: {0}.\n".format(
            cam.get_camera_information().serial_number))
    except Exception as e:
        print(e)

def load_image_into_numpy_array(image):
    try:
        ar = image.get_data()
        print(ar.shape)
        ar = ar[:, :, 0:3]
        (im_height, im_width, channels) = image.get_data().shape
    except Exception as e:
        print("Errror", e)
        raise e
    return np.array(ar).reshape((im_height, im_width, 3)).astype(np.uint8)

def main():
    cam, runtime = run_zed_camera()
    try:
        while True:
            frame = get_frame(cam, runtime)
            cv2.imshow("TestZedCamera", frame)
            key = cv2.waitKey(10)
            # if key == 'q':
            #     break
            # else:
            #     print(key)
        print("Wee")
        cam.close()
    except Exception as e:
        print(e)
        cam.close()

if __name__ == '__main__':
    main()
```

Sometimes the camera gives me one or two image then it gives this Error when I grab a new data from the camera
`err = cam.grab(runtime)`
I couldn't find the reason of this error. Before I was using the camera on a Windows machine and it didn't give me any issue.

adujardin, 2019-06-11:
Hi, 
This error code simply means that the computation was too fast, and that the new frame didn't change from the last grab call.
It's more like a warning. If this error occurs you should just add a small sleep and call the grab function again to get a new image.

zaher88abd, 2019-06-11:
Thanks for the fast reply.

closed 2019-06-11 by zaher88abd