stereolabs/zed-python-api #86: Disparity as Grayscale image

zaher88abd, 2019-06-20:
HI, 

I need the disparity as an image so what the best way to get it?
I am thinking to normalize the MEASURE_DISPARITY from 0-1 then multiply by 255 but the issue each image has a different range so it will be not accurate to make it in real time. Please, if a there better way to do it let me know.

adujardin, 2019-06-20:
Hi,
Yes, there is a better way to display the disparity map. Use the [`retrieveImage` function](https://github.com/stereolabs/zed-python-api/blob/master/examples/live_camera.py#L55) with  [`view=VIEW_DEPTH` ](https://www.stereolabs.com/docs/api/group__Video__group.html#ga77fc7bfc159040a1e2ffb074a8ad248c)

The disparity and depth are visually similar.

zaher88abd, 2019-06-20:
Ok, thank you @adujardin for this quick response.
I put [issue](https://github.com/stereolabs/zed-python-api/issues/79) from a long time and I will appreciate more if you could answer it.

closed 2019-06-20 by zaher88abd