stereolabs/zed-python-api #88: Positional Tracking stuck at "searching"

JosephRedfern, 2019-07-12:
Hello,

We recently (yesterday) collected a lot of data using a ZED Mini. On reviewing the data, the capture seemed to go well -- the depth data is quite clear, no exposure issues etc. We recorded the data using `zed.enable_recording()` and `zed.grab()` as per instructions, in 720P/60FPS mode. 

However, I am now attempting to extract trajectory data from the SVO files, using the ZED SDK. I have tried both ZEDfu, and using the SDK directly -- and it just won't work. The positional data never gets returned, and the tracking state is stuck at "SEARCHING", from the first frame onward.

I have uploaded a short (uncompressed) snippet of video here: https://redfern.me/public/misc/fail_example.svo (a H265 compressed version is available here: https://redfern.me/public/misc/fail_example_h265.svo, but I'm not sure how well the conversion worked).

Any help would be really appreciated. We have collected hundreds of gigabytes of data, and went to significant effort to arrange collection, so it would be a massive shame to not be able to use these SVO files.

Thanks in advance,
Joe

nesnes, 2019-07-15:
Thank you for the sample SVO files.
We were able to reproduce the tracking issue you reported. During the playback, the positional tracking keeps reporting a SEARCHING state.

Your SVO file is now being analyzed by our tracking experts.

I'll keep you informed of any updates

nesnes, 2019-07-15:
We found that you can use `initParameters.camera_disable_imu = true;` as a temporary fix to get tracking results.
We're looking at the IMU data to understand why it prevents the position estimation in your specific SVO files.

JosephRedfern, 2019-07-15:
@nesnes Great, thank you -- I will try that fix and let you know how we get on.

JosephRedfern, 2019-07-15:
@nesnes camera_disable_imu doesn't seem to exist in pyzed, but setting "enable_imu_fusion" to false in TrackingParameters allows us to get VO only translation/rotation, so works as a temporary fix (albeit with a potential loss in accuracy).

Look forward to hearing about why IMU data breaks tracking for our data -- if you need any more example files please let me know (including the full-length 3-minute clips). 

Thank you for the assistance,
Joe

JosephRedfern, 2019-07-15:
@nesnes Sorry for the flood of replies, but we have just tried using ZED Explorer to capture some data. 

We found that with compression DISABLED, tracking is "Searching" (when IMU enabled). However when we enable compression, tracking seems to work. I'm not sure if this is because IMU data is scrubbed from the file when compression is enabled, but perhaps this information will serve as another piece to the puzzle...

nesnes, 2019-07-25:
Hi,

We've just released the ZED SDK 2.8.3 that fixes the issue you were facing.
The compression mode you were using was subject to a bug preventing IMU data to be properly recorded.

This means that the IMU data in your existing SVO files cannot be recovered. However new SVO from the ZED SDK 2.8.3 won't suffer from this issue.

You can read the release note of the 2.8.3 and download it from our website: https://www.stereolabs.com/developers/release/

JosephRedfern, 2019-07-25:
@nesnes OK great, thank you for the update -- so this was only an issue when compression was disabled during recording?

We will update SDK ASAP. Thanks again for your efforts in sorting this, your level of support has been really excellent.

nesnes, 2019-07-25:
Yes, this was only affecting the LOSSLESS compression, this is why you were having no issues with LOSSY SVO files. Both compressions are now functionnal in the 2.8.3.

(Notice that LOSSLESS doesn't mean no compression, it's actually compressed but without loss of information)

JosephRedfern, 2019-07-25:
OK great, will close this issue. Thanks again!

closed 2019-07-25 by JosephRedfern