stereolabs/zed-python-api #94: Trouble in getting the color information from point cloud data

QijiaShao, 2019-08-09:
The documentation indicates that the point cloud stores its data using 32-bit float for each channel but when I print out the type, it is float 64. And the first 3 number which are the x,y,z coordinates are ok but the last data(color information) is weird.

Could anyone tell me how to get the right color information from zed camera? Thanks

fgtoralesch, 2019-08-23:
The point cloud is storing x,y,z,rgba if you retrieve it with `sl.MEASURE.MEASURE_XYZRGBA`. Using `point_cloud.get_data()` you will get an array with the 4 values for every pixel from the measurement. What you want to do then is interpret the float64 (8 x 8 bit values) as 4 uint8 values. So you could do the following:
`np.ravel(point_cloud.get_data()[:, :, 3])).view('uint8').reshape((720, 1280, 4))`
"Ravel" will arrange the point_cloud to be read as a 1-dimensional array in C format. "View" will reinterpret the float as 4 uint8 values and reshape will bring you back to having those 4 values assigned to every pixel. If you want to visualize the results with opencv you could just `[..., (2, 1, 0, 3)]` the array (since opencv assumes you have an bgra image and not rgba). 
I am not sure if this is what you wanted, but otherwise I am confident that you can get to the values that you want with the info.

saratrajput, 2020-03-11:
@fgtoralesch could you please expand on how to visualize the pointcloud? I've been trying to find an easy way to do a non blocking visualization in python, but haven't been able to. So far I've tried with open3d and pptk.

fennecinspace, 2022-04-26:
To add to @fgtoralesch, this is how to get the full data : X Y Z R G B A

```

# XYZ
xyz = point_cloud.get_data()[:, :, 0:3]

rgb = np.ravel(point_cloud.get_data()[:, :, 3]).view('uint8').reshape((h, w, 4))

rgba = [:,:,:3]

xyzrgb = np.concatenate((xyz, rgb), axis = -1)

xyzrgb = np.concatenate((xyz, rgba), axis = -1)
```

Thanks @fgtoralesch  this really helped me !

closed 2022-06-02 by github-actions[bot]