stereolabs/zed-python-api #95: ImportError: libhidapi-libusb.so.0

Muthukumar-S1979, 2019-08-09:
I have my ZED camera connected and i have installed ZED SDK with this script "ZED_SDK_Ubuntu16_v2.8.3.run" for my ubuntu-v16.4.

Installation completed successfully , while i try to run the samples "live_camera.py" from  zed-python-api-master/examples
Am getting below error

Traceback (most recent call last):
  File "live_camera.py", line 27, in <module>
    import pyzed.sl as sl
ImportError: libhidapi-libusb.so.0: cannot open shared object file: No such file or directory

Seems libhid and libusd.so drivers not installed properly from ZED SDK . Request to help me on this

obraun-sl, 2019-08-14:
Hi,
You can install hidapi packet with : 
`$ sudo apt-get install libhidapi-dev` 

That will install the missing libraries.

wangjunfeng90214, 2019-10-23:
When I try sudo apt-get install libhidapi-dev in Ubuntu 16.04, the terminal appears a picture like this:
[image]

What should I do to next?

obraun-sl, 2019-10-23:
Press Tab or your keyboard arrows so that the <OK> text is underligned. Then press OK.

closed 2022-04-27 by github-actions[bot]