stereolabs/zed-python-api #104: Problem during import pyzed.sl

gdubrg, 2019-09-18:
I have already installed Zed Python API through the `pip3` command.

In Python 3.6, `import pyzed` works well, but the `import pyzed.sl` gives the following error:
```
/usr/bin/python3.6: Relink `/lib/x86_64-linux-gnu/libsystemd.so.0' with `/lib/x86_64-linux-gnu/librt.so.1' for IFUNC symbol `clock_gettime'
/usr/bin/python3.6: Relink `/lib/x86_64-linux-gnu/libudev.so.1' with `/lib/x86_64-linux-gnu/librt.so.1' for IFUNC symbol `clock_gettime'
```
What can I do?

Poutsh, 2019-10-16:
Hi @gdubrg  ,

Have you tried using
```
import pyzed.sl as sl
```

If that doesn't work, have you tried building the python api from source :
https://github.com/stereolabs/zed-python-api#build-the-plugin-from-source

hzm2016, 2020-01-30:
I have the same issue, how can u solve it? many thanks

closed 2022-04-27 by github-actions[bot]