stereolabs/zed-python-api #106: Method to get the collision data through Depth information

Muthukumar-S1979, 2019-10-04:
Hi ,
Since we are getting depth information of whole scene, is there any way to get the "collision" information when two objects that collide with each other.
For example : can we able to identify whether the base ball collide with the human body using depth information. Another example , can we able to determine the golf ball collision with gold stick, etc..
Python sample to get this information  scenario would appreciable.

Advance thanks,
Muthukumar

Muthukumar-S1979, 2019-10-05:
Hi ,
Can any one help me on my query please.

Regards,
Muthukumar

obraun-sl, 2019-10-08:
Hi,
Collision detection will requires additionnal algorithms. The idea behind would be : 
1- to detect in 2D image the interesting object (human, golf ball.....)
2. Use the depth to transform the 2D bounding box of the object into 3D bounding box. 
3. Create an collider algorithm to detect if two 3D-BBox are overlapping (means collision).

Some interesting repos for that : 
https://github.com/stereolabs/zed-yolo --> the YOLO object detection , compatible with the ZED with advanced feature such as 3D position. 
https://github.com/flexible-collision-library/fcl --> never tested it on mu own but looks promising for collision detection (C++ however).

closed 2022-04-27 by github-actions[bot]