stereolabs/zed-python-api #107: Segmentation fault when enabling recording to file

mminervini, 2019-10-14:
Hi,

I get a segmentation fault when trying to enable recording to file on a Jetson Nano.
In particular, the following minimal example:
```
import pyzed.sl as sl
zed = sl.Camera()
init_params = sl.InitParameters()
status_open = zed.open(init_params)
if status_open == sl.ERROR_CODE.SUCCESS:
    status_recording = zed.enable_recording('test.svo', compression_mode=sl.SVO_COMPRESSION_MODE.SVO_COMPRESSION_MODE_HEVC)
    print(status_recording)
zed.close()
```
will crash at `enable_recording()` with the following error message:
```
Failed to query video capabilities: Inappropriate ioctl for device
Segmentation fault (core dumped)
```
Setting `compression_mode=sl.SVO_COMPRESSION_MODE.SVO_COMPRESSION_MODE_AVCHD` produces the same outcome. Instead, setting `compression_mode=sl.SVO_COMPRESSION_MODE.SVO_COMPRESSION_MODE_LOSSY` does not result in a crash.

It appears it has something to do with video encoding modes, however I cannot figure out which system library version (if any) does not get along with the ZED SDK.

I am working on a Jetson Nano with L4T 32.2.0, Cuda 10.0.326, Python 3.6.8, ZED SDK JP4.2 v2.8.3 and ZED Python API built from up-to-date source.
Please let me know if more specs are necessary to debug the issue.

obraun-sl, 2019-10-16:
Hi,
On jetson, AVCHD or HEVC uses the hardware encoder provided with tegra_multimedia_api installed by the Jetpack. Maybe something has changed in the last Jetpack.
Did you try with a previous jetpack for nano? I will try to reproduce this segfault

mminervini, 2019-10-17:
I grabbed and flashed the latest JetPack that was available few weeks ago when I was setting up the Nano, namely the jetson-nano-sd-r32.2-2019-07-16 SD card image.
Afterwards, the system has been upgraded with apt and several libraries have been installed, including audio-video libraries that were required to compile OpenCV (e.g., ffmpeg 3.4.6 and GStreamer 1.14.5). Are there any known conflicts with system libraries?

mminervini, 2019-10-19:
Quick update...
I re-flashed the SD card with the same image I used before (L4T R32.2) and only installed the minimum requirements to be able to record from the ZED cameras (i.e. ZED SDK, Python API, and a generic apt dist-upgrade). Now recording works!

It looks like there was some other system library conflicting with the ZED SDK and causing the segfault when trying to record.
With respect to current status, in the previous attempt I also installed the requirements to compile OpenCV and some other stuff...so, not sure what library is to blame.
@obraun-sl please let me know if you have any insights on this regard, just to know what I should avoid installing if I don't want to break the ZED capability of recording.

mminervini, 2019-10-24:
I get a segmentation fault also when loading an SVO file. Looks like there is an all-around issue with video encoding/decoding on Jetson platforms.

I am doing some tests on a slightly different platform than the original post, a Jetson Xavier with L4T 32.2.1, and in addition to getting a segfault when executing `enable_recording()`, I realized that the same error is triggered if I upload an SVO file on the Xavier and run this simple code to open it:
```
import pyzed.sl as sl
zed = sl.Camera()
init_params = sl.InitParameters(svo_input_filename='test.svo')
status = zed.open(init_params)
```
With the Xavier, the operating system gets installed with the NVIDIA SDK Manager, so I'm not sure if it is possible to try an older L4T release (as it was suggested by @obraun-sl).
On the other hand, I need to compile OpenCV with CUDA support, so I need to install a bunch of other libraries and it's hard to tell which one might be conflicting with the ZED SDK, without knowing its internals.

I'm a bit stuck with using ZED cameras on Jetson platforms...I look forward to any updates from the ZED team!
Thanks!

obraun-sl, 2019-10-24:
Hi,
To make sure I correctly understand, the recording+decoding issue happens only if OpenCV with CUDA is built and installed on the Jetson( Nano or Xavier), right ?
Basically the encoding/decoding of AVCHD/HEVC on jetson uses the library from tegra multimedia api (installed by the jetpack), v4l2-tegra and nvbuf_utils.

To make sure we are on the same platform, can you post the output of 
`$head -n 1 /etc/nv_tegra_release`
On my Xavier it gives : 
`# R32 (release), REVISION: 2.0, GCID: 15966166, BOARD: t186ref, EABI: aarch64, DATE: Wed Jul 17 00:26:04 UTC 2019`

obraun-sl, 2019-10-24:
Yes if you have the list , I can try to rebuild opencv with CUDA on my Xavier to check what's wrong

mminervini, 2019-10-24:
Looking at apt history, this should be the list of the packages I installed on the Xavier to build OpenCV 4.1.2: `build-essential cmake pkg-config git python3-pip python3-dev python3-tk graphviz libgraphviz-dev libgtk2.0-dev python-dev libgtk-3-dev libatlas3-base libblas-dev liblapack-dev liblapacke-dev libamd2 libeigen3-dev gfortran libtbb2 libtbb-dev libhdf5-dev libjpeg-dev libpng-dev libtiff-dev libtiff5-dev libdc1394-22-dev libgphoto2-dev libavcodec-dev libavformat-dev libswscale-dev libavresample-dev libv4l-dev v4l-utils qv4l2 v4l2ucp libxvidcore-dev libx264-dev libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev`

jinshengye-git, 2020-03-13:
I have the samilar problem 
I am launching  `roslaunch zed_wrap zed_camera.launch` with  debug mode,
when I try to use `rosservice call /zed_node/start_remote_stream` I got error like 

```
zed_wrapper_nod" received signal SIGSEGV Segmentation fault
Switching to Thread 0x7fadffb180 (LWP 22040)]
0x0000007f8006d150 in ?? () from /usr/local/zed/lib/libsl_svo.so
(gdb)
```

and  I tried with different zed sdk version error still there,
I once tried to re-make the /usr/src/tegra_multimedia_api/samples/12_camera_v4l2_cuda/camera_v4l2_cuda.cpp

I think I destroyed some link of the api application with libsl_svo.so

is there any solutions?

on my xavier:
```
head -n 1 /etc/nv_tegra_release                              
# R32 (release), REVISION: 2.0, GCID: 15966166, BOARD: t186ref, EABI: aarch64, DATE: Wed Jul 17 00:26:04 UTC 2019
```

mminervini, 2020-06-30:
Hi @obraun-sl,

I was wondering if there is any update on this issue, that I encountered again, now on a Jetson Xavier with L4T R32 revision 3.1, ZED SDK 3.2.0, and the latest version of the ROS wrapper.

The minimal Python example in the original post (slightly revised to conform to the new API) still results in a segmentation fault.
Video recording also fails when running the ZED ROS wrapper, with additional logs that I include below in case they provide further details that might help solve this issue.

For example, if I start a `zed_wrapper_node`, it keeps running without glitches until I call the SVO recording service, e.g. `rosservice call /zed_node/start_svo_recording ~/Videos/test.svo`, which fails with this message:
```
ERROR: transport error completing service call: unable to receive data from sender, check sender's logs for details
```
At the same time, in the `roslaunch` console I get:
```
Failed to query video capabilities: Inappropriate ioctl for device
[NvV4l2Elements] failed to open
Error in VIDIOC_REQBUFS at output planeError in VIDIOC_REQBUFS at output plane[ZED][Encoding] Critical Error : No Video Enc
```

Thanks in advance!

avl4400Etrac, 2021-07-13:
Any updates on this topic? I also tried to record svo from ros-service call on a JetsonXavier with JP4.4 an get the error:
```
Failed to query video capabilities: Inappropriate ioctl for device
libv4l2: error getting capabilities: Inappropriate ioctl for device
[NvV4l2Elements] failed to open
Error in VIDIOC_REQBUFS at output planeError in VIDIOC_REQBUFS at output plane[ZED][Encoding] Critical Error : No Video Enc
```
BR

closed 2022-04-27 by github-actions[bot]