stereolabs/zed-python-api #111: Memory leak in SVO reader

rbrigden, 2019-11-14:
We are reading a dataset of SVOs in a loop as part of a batch processing job and noticed a significant memory leak. In one specific application, we seek to random positions in the SVO before grabbing frames. In this case, memory usage grows unbounded at a rate of around 2.5 MBps until our host crashes.

Here is a plot of memory usage by a single process reading a set of SVOs (one at a time) and seeking to random frames in each SVO. 

[image]

We were able to reproduce this memory leak using a snippet from the `read_svo.py` example, although the memory does not leak as rapidly.

```python
import pyzed.sl as sl

def main():

    path = "/path/to/video.svo"
    print("Reading SVO file: {0}".format(path))

    init = sl.InitParameters(svo_input_filename=path ,svo_real_time_mode=False)
    cam = sl.Camera()
    status = cam.open(init)
    if status != sl.ERROR_CODE.SUCCESS:
        print(repr(status))
        exit()

    runtime = sl.RuntimeParameters()
    mat = sl.Mat()
    while True:  # for 'q' key
        err = cam.grab(runtime)
        if err == sl.ERROR_CODE.SUCCESS:
            cam.retrieve_image(mat)
        else:
            break
    cam.close()

if __name__ == "__main__":
    for _ in range(40):
        main()
```

Here is the memory usage by the process over time for the above program

[image]

These programs are run with pyzed 2.8 on Ubuntu 18 using a RTX2080.

adujardin, 2019-11-18:
Thank you for reporting this issue, we'll investigate and keep you posted.

closed 2020-04-09 by rbrigden

AlexanderKhazatsky, 2023-02-10:
Hello, im experiencing the same issue. Was this resolved?