stereolabs/zed-python-api #113: Anomalies during recording with H265

JosephRedfern, 2019-11-19:
Hello,

I am in the process of collecting some data using a ZED Camera, now using H265 for recording.

About a third of the time, this works -- in 720P/60Hz mode, a 30s clip ends up ~30mb. However, sometimes, the same 30s recording is x10-15 times that size. When opening in ZED Explorer, the  larger SVO files are reported to be well below the 60FPS requested, and when opening using the Python API, it hangs indefinitely and I have to kill the Python process, making it impossible to pass through our processing pipeline.

An example of a failed recording is here: https://redfern.me/public/failed_file.svo, with a successful  example available here: https://redfern.me/public/ok_file.svo.

Do you have any idea what might be causing this issue, or have any ideas how it might be fixed?

The files were recorded under Ubuntu, on a machine with an i7 7700T, GTX 1070 and a 1tb Samsung 960 Evo SSD.

obraun-sl, 2019-11-19:
Hi,

Thanks for reporting this issue. 
There is indeed a mismatch between the FPS and the timestamps of each frame in the failed SVO. When reading the failed_svo with ZED Explorer, the FPS is about 5fps which obviously is wrong when you look at the images. 
Do you use ZED Explorer or a recording sample (C++ or python) to save the SVO file ? Just to know where to look for the fix.
We will investigate on that issue.

Right now, to fix the failed SVO, there is a trick to recompress/restore the svo file. This idea is to use ZED SVOEditor with a correct svo (ok_file.svo) and a wrong svo (failed_file.svo) to "fake" the mismatch between fps and timestamps. Here is how to do : 

1) Save the number of frames for the "ok" svo : 
`$ ./ZED\ SVOEditor -inf ok_file.svo`
For example, on the "ok_file.svo" you shared, the number of frame is n=1741. 

2) Create a single svo that merge ok and failed (in that order)
`$ ./ZED\ SVOEditor --merge ok_fle.svo failed_file.svo temp_output.svo`

3) Cut the merged svo to extract only the "failed" part that start at s = number of frame +1
`$ ./ZED\ SVOEditor -cut temp_output.svo -s 1742 failed_file_restore.svo`

failed_file_restore.svo should be now at the proper size (around 60Mb) + with correct FPS.
make sure that ok_file and failed_file are in the same resolution ,otherwise it won't work.

JosephRedfern, 2019-11-19:
Thank you, I will try this out. 

The recording was done using the Python SDK Bindings, with enable_recording().

Are you able to reproduce the crash/hang while trying to read in the failed SVO with Python?

If you need me to share any code then please let me know. 

Thanks,
Joe

JosephRedfern, 2019-11-20:
Your process seems to work as temporary workaround -- thank you. I wrapped it up into a bash script, in case anyone else has this issue:

```
#!/bin/bash
set -e

OK_FILE=svo_1574168934.svo

if [ "$#" -ne 1 ]; then
    echo "Illegal number of parameters"
    exit 1
fi

broken_file=$1
fixed_file="fixed_"$broken_file

ok_frame_count=$(ZED\ SVOEditor -inf $OK_FILE 2>&1 |grep 'Number of Frames' | awk '{print $5'})

echo "[i] $ok_frame_count frames in OK file"

merged_file=$(mktemp /tmp/zed_merged_tmp.XXXXXXXX)
broken_start_index=$(($ok_frame_count + 1))

echo "[i] Starting merge..."

echo ZED\ SVOEditor --merge $OK_FILE $broken_file $merged_file
ZED\ SVOEditor --merge $OK_FILE $broken_file $merged_file
echo "[+] Merge complete, cropping"

ZED\ SVOEditor -cut $merged_file -s $broken_start_index $fixed_file

echo "[+] Crop completed"
echo "[i] Deleting temporary file"

rm $merged_file

echo "Fixed file written to $fixed_file"

``` 

`ok_file` needs to be set to a known-good SVO, and `ZED\ SVOEditor` needs to be on your PATH.

JosephRedfern, 2019-12-12:
@obraun-sl Any updates on this issue? I've seen it happening in SVO Explorer too, and it's quite inconvenient having to do the additional processing, and the corrupt files occupy a large amount of disk space.

closed 2022-04-27 by github-actions[bot]