stereolabs/zed-python-api #120: Segmentation Fault with Mat.get_data(memory_type=sl.MEM.MEM_GPU)

ajwillia, 2020-01-18:
I'm running on a Jetson TX2 development board with latest JetPack and latest ZED SDK.  ZED camera is latest firmware.
python3 --version
Python 3.6.9

When I make a call to Mat.get_data(memory_type=sl.MEM.MEM_GPU) I get a Segmentation Fault.

Code to repro:

    import pyzed.sl as sl

    if __name__ == '__main__':
        zed = sl.Camera()
        
        init_params = sl.InitParameters()
        init_params.camera_resolution = sl.RESOLUTION.RESOLUTION_VGA
        init_params.depth_mode = sl.DEPTH_MODE.DEPTH_MODE_PERFORMANCE

        # Open the camera
        status = zed.open(init_params)
        if status != sl.ERROR_CODE.SUCCESS:
            print(repr(status))
            exit()	    

        resolution = zed.get_resolution()
        # create matrices for camera image and depth info in GPU memory
        image = sl.Mat(resolution.width, 
                        resolution.height,
                        sl.MAT_TYPE.MAT_TYPE_8U_C4,
                        memory_type=sl.MEM.MEM_GPU)

        depth = sl.Mat(resolution.width, 
                            resolution.height,
                            sl.MAT_TYPE.MAT_TYPE_32F_C1,
                            memory_type=sl.MEM.MEM_GPU) 

        if zed.grab(runtime_parameters) == sl.ERROR_CODE.SUCCESS:     
            zed.retrieve_image(image, sl.VIEW.VIEW_LEFT,
                            type=sl.MEM.MEM_GPU)
            # Retrieve depth map. Depth is aligned on the left image
            zed.retrieve_measure(depth, sl.MEASURE.MEASURE_DEPTH,
                                    type=sl.MEM.MEM_GPU)	
            
            
            #SegFault!
            d = image.get_data(memory_type=sl.MEM.MEM_GPU)                        

Am I doing something wrong?  If not, is there a work-around?  I plan on using opencv cuda functions for image processing which is why I need the call to get_data in GPU memory

Poutsh, 2020-01-20:
Hi @ajwillia ,

The GPU get_data is currently not supported in our python API because it can not be converted in numpy arrays without additional dependencies. We'll be exploring this and might end up implementing a dedicated sample in an upcoming version of the ZED SDK.
For now we recommend you to use CPU memories in Python, however, if performances are crucial and you do need GPU optimizations, you can use our [C++ API](https://github.com/stereolabs/zed-examples).

ajwillia, 2020-01-20:
Hi @Poutsh 

Thanks for the reply.  I can get this to work by making a call to update_cpu_from_gpu()

        if zed.grab(runtime_parameters) == sl.ERROR_CODE.SUCCESS:     
            zed.retrieve_image(image, sl.VIEW.VIEW_LEFT,
                            type=sl.MEM.MEM_GPU)
            # Retrieve depth map. Depth is aligned on the left image
            zed.retrieve_measure(depth, sl.MEASURE.MEASURE_DEPTH,
                                    type=sl.MEM.MEM_GPU)	

            # avoid segfault by copying mat to CPU memory
            image.update_cpu_from_gpu()
            d = image.get_data()           

As you said, I may develop in c++ if there is too much of a performance penalty copying from GPU to CPU.

closed 2020-01-20 by ajwillia

akila-ganlath-tm, 2022-01-20:
echo @skuley  request

lafernandes22, 2022-02-22:
Also curious if there's an update on this too

romanwicky, 2022-02-22:
This bottleneck would be great to improve on. @abastie any updates?