stereolabs/zed-python-api #122: libsl_core.so Error ?

MohamedAliRashad, 2020-01-26:
i get this error everytime i try to import the ```pyzed.sl``` module

```
libsl_core.so: cannot open shared object file: No such file or directory
```

Poutsh, 2020-01-27:
Hi @MohamedAliRashad ,

Can you tell me which version of the ZED SDK are you using?

MohamedAliRashad, 2020-01-27:
@Poutsh 
ZED SDK 3.0 with cuda 10.2

Poutsh, 2020-01-27:
@MohamedAliRashad 

Have you rebuilt/reinstalled the ZED Python API after you installed ZED SDK 3.0?
Can you give me the output of :
```
ls /usr/local/lib/python3.6/dist-packages/pyzed*
```
Replace python3.6 with your version of python 3

MohamedAliRashad, 2020-01-27:
@Poutsh 
Nothing Changed.
Here is the output
[image]

Poutsh, 2020-01-27:
Is there a ```/usr/local/zed_previous``` directory in your system?
Can you delete it and rebuild/reinstall the python wrapper ?

MohamedAliRashad, 2020-01-28:
@Poutsh 
Still not working

Poutsh, 2020-01-28:
@MohamedAliRashad 
It looks like there is a pyzed compiled with SDK 2.8 that is present on your system (libsl_core.so was removed in 3.0)

You can install "locate" on Ubuntu to search for pyzed installs that can produce this behavior :
```shell
sudo apt-get install locate
sudo updatedb
locate pyzed
```

MohamedAliRashad, 2020-01-28:
@Poutsh 
you were correct
[image]

How to remove them ?
i tried ```sudo python3 setup.py cleanall``` but it only removed the 3.0 files

Poutsh, 2020-01-29:
@MohamedAliRashad 
You can remove those files using the rm command
```
rm -rf .local/lib/python3.6/site-packages/pyzed
rm -rf .local/lib/python3.6/site-packages/pyzed-2.8.dist-info
```

MohamedAliRashad, 2020-01-29:
@Poutsh 
I had to use the sudo command because of this error
[image]

Poutsh, 2020-01-29:
@MohamedAliRashad 
Yes without the --user option you have to use sudo.
Have your problem been solved ?

MohamedAliRashad, 2020-01-29:
@Poutsh 
Yes, Thanks Alot.

closed 2020-01-29 by MohamedAliRashad