stereolabs/zed-python-api #123: build setup.py errors in camera.hpp in ZED SDK

majinshaoyuindustry, 2020-01-27:
Hello all,

I am running on anaconda 3 powerShell with Zed SDK 3.0, VS 2015. Installed requirements in the env called "ten". Here is my errors:

(base) PS C:\WINDOWS\system32> d:
(base) PS D:\> cd zed-python-api
(base) PS D:\zed-python-api> conda activate ten
(ten) PS D:\zed-python-api> python setup.py cleanall
Deleting Cython files ..
(ten) PS D:\zed-python-api> python setup.py build
ZED SDK Version: OK
compilation flags:
include dirs: ['C:\\Users\\ssu\\Anaconda3\\envs\\ten\\lib\\site-packages\\numpy\\core\\include', 'C:\\Program Files (x86)\\ZED SDK/include', 'C:\\Program Files\\NVIDIA GPU Computing Toolkit\\CUDA\\v10.2/include']
library dirs: ['C:\\Users\\ssu\\Anaconda3\\envs\\ten\\lib\\site-packages\\numpy\\core\\include', 'C:\\Program Files (x86)\\ZED SDK/lib', 'C:\\Program Files\\NVIDIA GPU Computing Toolkit\\CUDA\\v10.2/lib/x64']
libraries: ['sl_zed64']
Building module: ('pyzed.sl', ['pyzed/sl.pyx'])
running build
running build_py
creating build
creating build\lib.win-amd64-3.7
creating build\lib.win-amd64-3.7\pyzed
copying pyzed\__init__.py -> build\lib.win-amd64-3.7\pyzed
running build_ext
building 'pyzed.sl' extension
creating build\temp.win-amd64-3.7
creating build\temp.win-amd64-3.7\Release
creating build\temp.win-amd64-3.7\Release\pyzed
C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\BIN\x86_amd64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -I.\pyzed -IC:\Users\ssu\Anaconda3\envs\ten\lib\site-packages\numpy\core\include "-IC:\Program Files (x86)\ZED SDK/include" "-IC:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v10.2/include" -IC:\Users\ssu\Anaconda3\envs\ten\include -IC:\Users\ssu\Anaconda3\envs\ten\include "-IC:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\INCLUDE" "-IC:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\ATLMFC\INCLUDE" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\winrt" /EHsc /Tppyzed/sl.cpp /Fobuild\temp.win-amd64-3.7\Release\pyzed/sl.obj
sl.cpp
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(4375): error C2146: syntax error: missing ';' before identifier 'getZEDSDKRuntimeVersion'
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(4375): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(4406): error C2182: 'SL_SDK_EXPORT': illegal use of type 'void'
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(4406): error C2433: 'SL_SDK_EXPORT': 'inline' not permitted on data declarations
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(4406): error C2086: 'int SL_SDK_EXPORT': redefinition
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(4375): note: see declaration of 'SL_SDK_EXPORT'
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(4406): error C2146: syntax error: missing ';' before identifier 'getZEDSDKBuildVersion'
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(4406): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(4410): warning C4508: 'getZEDSDKBuildVersion': function should return a value; 'void' return type assumed
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(4460): error C2146: syntax error: missing ';' before identifier 'toString'
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(4460): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(4492): error C2086: 'sl::String sl::SL_SDK_EXPORT': redefinition
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(4460): note: see declaration of 'sl::SL_SDK_EXPORT'
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(4492): error C2146: syntax error: missing ';' before identifier 'toString'
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(4492): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(4513): error C2086: 'sl::String sl::SL_SDK_EXPORT': redefinition
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(4460): note: see declaration of 'sl::SL_SDK_EXPORT'
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(4513): error C2146: syntax error: missing ';' before identifier 'toString'
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(4513): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(4534): error C2086: 'sl::String sl::SL_SDK_EXPORT': redefinition
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(4460): note: see declaration of 'sl::SL_SDK_EXPORT'
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(4534): error C2146: syntax error: missing ';' before identifier 'toString'
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(4534): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(4568): error C2086: 'sl::String sl::SL_SDK_EXPORT': redefinition
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(4460): note: see declaration of 'sl::SL_SDK_EXPORT'
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(4568): error C2146: syntax error: missing ';' before identifier 'toString'
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(4568): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(4599): error C2086: 'sl::String sl::SL_SDK_EXPORT': redefinition
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(4460): note: see declaration of 'sl::SL_SDK_EXPORT'
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(4599): error C2146: syntax error: missing ';' before identifier 'toString'
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(4599): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(4618): error C2086: 'sl::String sl::SL_SDK_EXPORT': redefinition
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(4460): note: see declaration of 'sl::SL_SDK_EXPORT'
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(4618): error C2146: syntax error: missing ';' before identifier 'toString'
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(4618): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(4639): error C2086: 'sl::String sl::SL_SDK_EXPORT': redefinition
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(4460): note: see declaration of 'sl::SL_SDK_EXPORT'
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(4639): error C2146: syntax error: missing ';' before identifier 'toString'
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(4639): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(4662): error C2086: 'sl::String sl::SL_SDK_EXPORT': redefinition
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(4460): note: see declaration of 'sl::SL_SDK_EXPORT'
[... 99 lines cut ...]
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(5336): error C2550: 'void __cdecl `dynamic initializer for 'enable_pose_smoothing''(void)': constructor initializer lists are only allowed on constructor definitions
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(5336): error C2065: '_area_path': undeclared identifier
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(5336): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(5337): error C2550: 'void __cdecl `dynamic initializer for 'area_file_path''(void)': constructor initializer lists are only allowed on constructor definitions
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(5337): error C2065: '_set_floor_as_origin': undeclared identifier
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(5337): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
[... cut ...]
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(5339): error C2448: 'set_as_static': function-style initializer appears to be a function definition
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(5347): error C2146: syntax error: missing ')' before identifier 'filename'
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(5347): fatal error C1003: error count exceeds 100; stopping compilation
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio 14.0\\VC\\BIN\\x86_amd64\\cl.exe' failed with exit status 2
(ten) PS D:\zed-python-api> 

Has anyone encountered the same issues? Please give any insights, thank you!

majinshaoyuindustry, 2020-01-27:
Update: I created a new conda env with Numpy 1.13.1 and Cython 0.26 (exactly as in the tutorial for ZED SDK 3.0) Now I got:

(base) C:\WINDOWS\system32>conda activate zedsdk

(zedsdk) C:\WINDOWS\system32>d:

(zedsdk) D:\>cd zed-python-api

(zedsdk) D:\zed-python-api>python setup.py build
ZED SDK Version: OK
compilation flags:
include dirs: ['C:\\Users\\ssu\\Anaconda3\\envs\\zedsdk\\lib\\site-packages\\numpy\\core\\include', 'C:\\Program Files (x86)\\ZED SDK/include', 'C:\\Program Files\\NVIDIA GPU Computing Toolkit\\CUDA\\v10.2/include']
library dirs: ['C:\\Users\\ssu\\Anaconda3\\envs\\zedsdk\\lib\\site-packages\\numpy\\core\\include', 'C:\\Program Files (x86)\\ZED SDK/lib', 'C:\\Program Files\\NVIDIA GPU Computing Toolkit\\CUDA\\v10.2/lib/x64']
libraries: ['sl_zed64']
Building module: ('pyzed.sl', ['pyzed/sl.pyx'])
Compiling pyzed/sl.pyx because it changed.
[1/1] Cythonizing pyzed/sl.pyx

Error compiling Cython file:
------------------------------------------------------------
...
## \defgroup Sensors_group Sensors Module

##
# \ref Timesamp representation and utilities.
# \ingroup Core_group
cdef class Timestamp():
                    ^
------------------------------------------------------------

pyzed\sl.pyx:49:21: Expected an identifier
Traceback (most recent call last):
  File "setup.py", line 183, in <module>
    extList = cythonize(extension, compiler_directives=cython_directives)
  File "C:\Users\ssu\Anaconda3\envs\zedsdk\lib\site-packages\Cython\Build\Dependencies.py", line 1027, in cythonize
    cythonize_one(*args)
  File "C:\Users\ssu\Anaconda3\envs\zedsdk\lib\site-packages\Cython\Build\Dependencies.py", line 1149, in cythonize_one
    raise CompileError(None, pyx_file)
Cython.Compiler.Errors.CompileError: pyzed/sl.pyx

(zedsdk) D:\zed-python-api>

adujardin, 2020-01-27:
Could you make sure you took the latest master branch (and clear any cache) which includes the support of 3.0? It seems it may be related to a mismatch of versions between the SDK and the python wrapper.

majinshaoyuindustry, 2020-01-27:
> Could you make sure you took the latest master branch (and clear any cache) which includes the support of 3.0? It seems it may be related to a mismatch of versions between the SDK and the python wrapper.

Thank you for your reply. I downloaded the api from here: https://github.com/stereolabs/zed-python-api
And I used to have ZED sdk 2.8x then updated to 3.0 today. Could you please further explain the method to clear cache/taking lastest master branch?

Thank you!

majinshaoyuindustry, 2020-01-27:
One thing I noticed: 
In here (https://github.com/stereolabs/zed-python-api) it asked for Cython 0.26
But in requirement.txt it asked for Cython>= 0.28
Let me know if this even matters.

adujardin, 2020-01-27:
It may not be the same problem but we found an issue that prevents from compiling the wrapper on Windows, due to a flag being incorrectly interpreted. 

We'll publish a fix soon, if you want to test it you can replace `#if defined WIN32` by `#if defined _WIN32` in `C:\Program Files (x86)\ZED SDK\include\sl\Camera.hpp` on line 4348

majinshaoyuindustry, 2020-01-27:
> It may not be the same problem but we found an issue that prevents from compiling the wrapper on Windows, due to a flag being incorrectly interpreted.
>
> We'll publish a fix soon, if you want to test [...]

Thank you.
Performed: "replace #if defined WIN32 by #if defined _WIN32 in C:\Program Files (x86)\ZED SDK\include\sl\Camera.hpp on line 4348"

Cython=0.28 (0.26 can't even reach to camera.hpp)

And here it is:
(zedsdk) D:\zed-python-api>python setup.py cleanall
Deleting Cython files ..

(zedsdk) D:\zed-python-api>python setup.py build
ZED SDK Version: OK
compilation flags:
include dirs: ['C:\\Users\\ssu\\Anaconda3\\envs\\zedsdk\\lib\\site-packages\\numpy\\core\\include', 'C:\\Program Files (x86)\\ZED SDK/include', 'C:\\Program Files\\NVIDIA GPU Computing Toolkit\\CUDA\\v10.2/include']
library dirs: ['C:\\Users\\ssu\\Anaconda3\\envs\\zedsdk\\lib\\site-packages\\numpy\\core\\include', 'C:\\Program Files (x86)\\ZED SDK/lib', 'C:\\Program Files\\NVIDIA GPU Computing Toolkit\\CUDA\\v10.2/lib/x64']
libraries: ['sl_zed64']
Building module: ('pyzed.sl', ['pyzed/sl.pyx'])
running build
running build_py
creating build
creating build\lib.win-amd64-3.6
creating build\lib.win-amd64-3.6\pyzed
copying pyzed\__init__.py -> build\lib.win-amd64-3.6\pyzed
running build_ext
building 'pyzed.sl' extension
creating build\temp.win-amd64-3.6
creating build\temp.win-amd64-3.6\Release
creating build\temp.win-amd64-3.6\Release\pyzed
C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\BIN\x86_amd64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -I.\pyzed -IC:\Users\ssu\Anaconda3\envs\zedsdk\lib\site-packages\numpy\core\include "-IC:\Program Files (x86)\ZED SDK/include" "-IC:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v10.2/include" -IC:\Users\ssu\Anaconda3\envs\zedsdk\include -IC:\Users\ssu\Anaconda3\envs\zedsdk\include "-IC:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\INCLUDE" "-IC:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\ATLMFC\INCLUDE" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\winrt" /EHsc /Tppyzed/sl.cpp /Fobuild\temp.win-amd64-3.6\Release\pyzed/sl.obj
sl.cpp
C:\Program Files (x86)\ZED SDK/include\sl/Camera.hpp(4348): warning C4067: unexpected tokens following preprocessor directive - expected a newline
c:\users\ssu\anaconda3\envs\zedsdk\lib\site-packages\numpy\core\include\numpy\npy_1_7_deprecated_api.h(12) : Warning Msg: Using deprecated NumPy API, disable it by #defining NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION
pyzed/sl.cpp(14295): error C3431: 'OBJECT_CLASS': a scoped enumeration cannot be redeclared as an unscoped enumeration
pyzed/sl.cpp(14362): error C3431: 'OBJECT_CLASS': a scoped enumeration cannot be redeclared as an unscoped enumeration
pyzed/sl.cpp(14429): error C3431: 'OBJECT_TRACKING_STATE': a scoped enumeration cannot be redeclared as an unscoped enumeration
pyzed/sl.cpp(14496): error C3431: 'OBJECT_TRACKING_STATE': a scoped enumeration cannot be redeclared as an unscoped enumeration
pyzed/sl.cpp(14563): error C3431: 'OBJECT_ACTION_STATE': a scoped enumeration cannot be redeclared as an unscoped enumeration
pyzed/sl.cpp(14630): error C3431: 'OBJECT_ACTION_STATE': a scoped enumeration cannot be redeclared as an unscoped enumeration
pyzed/sl.cpp(14942): error C3431: 'OBJECT_CLASS': a scoped enumeration cannot be redeclared as an unscoped enumeration
pyzed/sl.cpp(15158): error C3431: 'OBJECT_TRACKING_STATE': a scoped enumeration cannot be redeclared as an unscoped enumeration
pyzed/sl.cpp(15374): error C3431: 'OBJECT_ACTION_STATE': a scoped enumeration cannot be redeclared as an unscoped enumeration
pyzed/sl.cpp(17070): warning C4800: 'int': forcing value to bool 'true' or 'false' (performance warning)
pyzed/sl.cpp(17191): warning C4800: 'int': forcing value to bool 'true' or 'false' (performance warning)
pyzed/sl.cpp(17575): warning C4800: 'int': forcing value to bool 'true' or 'false' (performance warning)
pyzed/sl.cpp(17576): warning C4800: 'int': forcing value to bool 'true' or 'false' (performance warning)
pyzed/sl.cpp(17577): warning C4800: 'int': forcing value to bool 'true' or 'false' (performance warning)
pyzed/sl.cpp(17742): warning C4800: 'int': forcing value to bool 'true' or 'false' (performance warning)
pyzed/sl.cpp(17863): warning C4800: 'int': forcing value to bool 'true' or 'false' (performance warning)
pyzed/sl.cpp(17984): warning C4800: 'int': forcing value to bool 'true' or 'false' (performance warning)
pyzed/sl.cpp(23203): error C3431: 'MAT_TYPE': a scoped enumeration cannot be redeclared as an unscoped enumeration
pyzed/sl.cpp(23203): error C3431: 'MEM': a scoped enumeration cannot be redeclared as an unscoped enumeration
pyzed/sl.cpp(23377): error C3431: 'MAT_TYPE': a scoped enumeration cannot be redeclared as an unscoped enumeration
pyzed/sl.cpp(23377): error C3431: 'MEM': a scoped enumeration cannot be redeclared as an unscoped enumeration
pyzed/sl.cpp(23626): error C3431: 'MAT_TYPE': a scoped enumeration cannot be redeclared as an unscoped enumeration
pyzed/sl.cpp(23626): error C3431: 'MEM': a scoped enumeration cannot be redeclared as an unscoped enumeration
[... cut ...]
pyzed/sl.cpp(69429): warning C4800: 'int': forcing value to bool 'true' or 'false' (performance warning)
pyzed/sl.cpp(71363): warning C4800: 'int': forcing value to bool 'true' or 'false' (performance warning)
pyzed/sl.cpp(71746): warning C4800: 'int': forcing value to bool 'true' or 'false' (performance warning)
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio 14.0\\VC\\BIN\\x86_amd64\\cl.exe' failed with exit status 2

(zedsdk) D:\zed-python-api>

Looking forward to the fix and thank you for doing that!

majinshaoyuindustry, 2020-01-28:
@adujardin 

Updated to ZED Sdk 3.0.1
Here is the errors:

(zedsdk) D:\zed-python-api>python setup.py build
ZED SDK Version: OK
compilation flags:
include dirs: ['C:\\Users\\ssu\\Anaconda3\\envs\\zedsdk\\lib\\site-packages\\numpy\\core\\include', 'C:\\Program Files (x86)\\ZED SDK/include', 'C:\\Program Files\\NVIDIA GPU Computing Toolkit\\CUDA\\v10.2/include']
library dirs: ['C:\\Users\\ssu\\Anaconda3\\envs\\zedsdk\\lib\\site-packages\\numpy\\core\\include', 'C:\\Program Files (x86)\\ZED SDK/lib', 'C:\\Program Files\\NVIDIA GPU Computing Toolkit\\CUDA\\v10.2/lib/x64']
libraries: ['sl_zed64']
Building module: ('pyzed.sl', ['pyzed/sl.pyx'])
Compiling pyzed/sl.pyx because it changed.
[1/1] Cythonizing pyzed/sl.pyx
C:\Users\ssu\Anaconda3\envs\zedsdk\lib\site-packages\Cython\Compiler\Main.py:369: FutureWarning: Cython directive 'language_level' not set, using 2 for now (Py2). This will change in a later release! File: D:\zed-python-api\pyzed\sl.pyx
  tree = Parsing.p_module(s, pxd, full_module_name)
running build
running build_py
creating build
creating build\lib.win-amd64-3.6
creating build\lib.win-amd64-3.6\pyzed
copying pyzed\__init__.py -> build\lib.win-amd64-3.6\pyzed
running build_ext
building 'pyzed.sl' extension
creating build\temp.win-amd64-3.6
creating build\temp.win-amd64-3.6\Release
creating build\temp.win-amd64-3.6\Release\pyzed
C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\BIN\x86_amd64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -I.\pyzed -IC:\Users\ssu\Anaconda3\envs\zedsdk\lib\site-packages\numpy\core\include "-IC:\Program Files (x86)\ZED SDK/include" "-IC:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v10.2/include" -IC:\Users\ssu\Anaconda3\envs\zedsdk\include -IC:\Users\ssu\Anaconda3\envs\zedsdk\include "-IC:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\INCLUDE" "-IC:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\ATLMFC\INCLUDE" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.17763.0\winrt" /EHsc /Tppyzed/sl.cpp /Fobuild\temp.win-amd64-3.6\Release\pyzed/sl.obj
sl.cpp
c:\users\ssu\anaconda3\envs\zedsdk\lib\site-packages\numpy\core\include\numpy\npy_1_7_deprecated_api.h(12) : Warning Msg: Using deprecated NumPy API, disable it by #defining NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION
pyzed/sl.cpp(13973): error C3431: 'OBJECT_CLASS': a scoped enumeration cannot be redeclared as an unscoped enumeration
pyzed/sl.cpp(14040): error C3431: 'OBJECT_CLASS': a scoped enumeration cannot be redeclared as an unscoped enumeration
pyzed/sl.cpp(14107): error C3431: 'OBJECT_TRACKING_STATE': a scoped enumeration cannot be redeclared as an unscoped enumeration
pyzed/sl.cpp(14174): error C3431: 'OBJECT_TRACKING_STATE': a scoped enumeration cannot be redeclared as an unscoped enumeration
pyzed/sl.cpp(14241): error C3431: 'OBJECT_ACTION_STATE': a scoped enumeration cannot be redeclared as an unscoped enumeration
pyzed/sl.cpp(14308): error C3431: 'OBJECT_ACTION_STATE': a scoped enumeration cannot be redeclared as an unscoped enumeration
pyzed/sl.cpp(14588): error C3431: 'OBJECT_CLASS': a scoped enumeration cannot be redeclared as an unscoped enumeration
pyzed/sl.cpp(14772): error C3431: 'OBJECT_TRACKING_STATE': a scoped enumeration cannot be redeclared as an unscoped enumeration
pyzed/sl.cpp(14956): error C3431: 'OBJECT_ACTION_STATE': a scoped enumeration cannot be redeclared as an unscoped enumeration
pyzed/sl.cpp(16662): warning C4800: 'int': forcing value to bool 'true' or 'false' (performance warning)
pyzed/sl.cpp(16783): warning C4800: 'int': forcing value to bool 'true' or 'false' (performance warning)
pyzed/sl.cpp(17167): warning C4800: 'int': forcing value to bool 'true' or 'false' (performance warning)
pyzed/sl.cpp(17168): warning C4800: 'int': forcing value to bool 'true' or 'false' (performance warning)
pyzed/sl.cpp(17169): warning C4800: 'int': forcing value to bool 'true' or 'false' (performance warning)
pyzed/sl.cpp(17334): warning C4800: 'int': forcing value to bool 'true' or 'false' (performance warning)
pyzed/sl.cpp(17455): warning C4800: 'int': forcing value to bool 'true' or 'false' (performance warning)
pyzed/sl.cpp(17576): warning C4800: 'int': forcing value to bool 'true' or 'false' (performance warning)
pyzed/sl.cpp(22787): error C3431: 'MAT_TYPE': a scoped enumeration cannot be redeclared as an unscoped enumeration
pyzed/sl.cpp(22787): error C3431: 'MEM': a scoped enumeration cannot be redeclared as an unscoped enumeration
pyzed/sl.cpp(22961): error C3431: 'MAT_TYPE': a scoped enumeration cannot be redeclared as an unscoped enumeration
pyzed/sl.cpp(22961): error C3431: 'MEM': a scoped enumeration cannot be redeclared as an unscoped enumeration
pyzed/sl.cpp(23207): error C3431: 'MAT_TYPE': a scoped enumeration cannot be redeclared as an unscoped enumeration
pyzed/sl.cpp(23207): error C3431: 'MEM': a scoped enumeration cannot be redeclared as an unscoped enumeration
pyzed/sl.cpp(23467): error C3431: 'MAT_TYPE': a scoped enumeration cannot be redeclared as an unscoped enumeration
pyzed/sl.cpp(23671): error C3431: 'MAT_TYPE': a scoped enumeration cannot be redeclared as an unscoped enumeration
[... cut ...]
pyzed/sl.cpp(66928): warning C4800: 'int': forcing value to bool 'true' or 'false' (performance warning)
pyzed/sl.cpp(68546): warning C4800: 'int': forcing value to bool 'true' or 'false' (performance warning)
pyzed/sl.cpp(68897): warning C4800: 'int': forcing value to bool 'true' or 'false' (performance warning)
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio 14.0\\VC\\BIN\\x86_amd64\\cl.exe' failed with exit status 2

(zedsdk) D:\zed-python-api>

Have you tried to build&install it on Win10? If so, could you please share your setups so I can clone it?
Thank you!

adujardin, 2020-01-28:
We tested on Windows without any issues, but it seems the new version of the wrapper needs at least **Python 3.7** to compile. This issue isn't present on Linux (Ubuntu 16 default python is 3.5 and it works fine). 
It's a combination of python version and compiler. We'll try to find a fix. In the meantime, you should update to Python 3.7 if possible.

adujardin, 2020-01-28:
The latest eb8b0ed269fa557508c4cd85497f3abdcae0fe6e commit should fix your issue, even on VS 2015 with python 3.5+

majinshaoyuindustry, 2020-01-29:
> The latest [eb8b0ed](https://github.com/stereolabs/zed-python-api/commit/eb8b0ed269fa557508c4cd85497f3abdcae0fe6e) commit should fix your issue, even on VS 2015 with python 3.5+

Thank you! I upgraded to python 3.7.6 and it all worked all. I will close this ticket.

closed 2020-01-29 by majinshaoyuindustry